prusa3d/PrusaSlicer #2693: Importing 3MF: Losing object position.

morimorida, 2019-07-30:
### Version
PrusaSlicer 2.0.0 201905201652 Win64

Windows 10

Prusa i3 MK2/Prusa i3 MK3S + MMU2S

### Behavior
When I import a 3MF file, the position of the object is lost and get imported as loose object.

1. Import STL file in Blender
2. Color it (separate selected surfaces and give different color)
3. Export it as Obj file with color information.
4. Import obj file in 3D builder.
5. Export it as 3MF.
6. Import it into PrusaSlicer

Workaround....
I did find a workaround just now to...:
1) Import the 3mf into Simplify 3D->Separate connected Surfaces
2) Export the individual stl files
3) Import all stl file into PrusaSlicer.

However, it would be nice if it works without using Simplify 3D.

[image]

bubnikv, 2019-07-30:
please attach 3mf

bubnikv, 2019-08-01:
I believe your issue is the same as #2317, see the comments in #2317. The issue will be fixed by the upcoming PrusaSlicer 2.1.0-alpha1.

closed 2019-08-01 by bubnikv