prusa3d/PrusaSlicer #2694: [FEATURE REQUEST] Add the option to "not print" a part

RossKor, 2019-07-30:
### Version
2.0.0+ win64

### Feature Request
Allow the user to toggle the print / do-not-print status of a part within a project.
This could live in the part sidebar context menu (see attached pic).
[image]

### Justification
I would like the option to print all, or only some (e.g. one or more parts failed, but not all), without worrying about accidentally saving my project (and overwriting it) after removing one or more parts.

neophyl, 2019-07-30:
You could just add a modifier over the part you don't want to print and set infill and all the perimeters to 0.  That stops the part printing.  Not as convenient as a tick box admittedly  but it does work.

enricoturri1966, 2019-07-31:
Working on it -> https://dev.prusa3d.com/browse/SPE-995

PRGeno, 2019-08-01:
Making it visible or invisible on the bed display would also be desirable.

YuSanka, 2019-08-09:
Thanks for your suggestion. This feature is implemented and will be a part of upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-09 by YuSanka