prusa3d/PrusaSlicer #2695: [Feature Request] Export STL w Modifiers

mineshaftgap, 2019-07-30:
It would be great to be able to have modifiers in PrusaSlicer export to STL for sharing with others. Gcode will have a infill modifier with a perimeter set to 0, but the STL will still be the same original object.

bubnikv, 2019-07-30:
modifiers are exported in 3MF to be shared with others.

zladson, 2019-08-07:
This option was some what implemented (at least in terms of supports) however, there seems to be an issue with it. File > Export > Export Plate as STL with supports. It seems that no matter what I do I am unable to use this option. It is always grayed out.  While the 3MF works for sure, why would this feature be implemented if the goal was to use 3MF? Am I missing something with the export feature? Should I open a new case?

bubnikv, 2019-08-08:
>  This option was some what implemented (at least in terms of supports)

That is for SLA supports only.

bubnikv, 2019-08-09:
Why do you need to export STL with modifiers? What is the purpose of such action?

zladson, 2019-08-09:
When creating a model to share, being able to export the model as an STL with custom supports set up via a slicer is a much easier/smoother method (IMO) then building supports into design itself, it also allows us to use the cool features of Plicer like the support interfacing.

rtyr, 2020-08-19:
I believe this issue was already answered. Object with custom modifiers can be exported to 3MF format. Closing.

closed 2020-08-19 by rtyr