prusa3d/PrusaSlicer #2696: [Build] Error when make PrusaSlicer

userosos, 2019-07-30:
Hi! I can not install the slicer:
1. Install Microsoft Visual Studio.

> Microsoft Visual Studio Enterprise 2019 (2)
> Version 16.2.0
> VisualStudio.16.Release/16.2.0+29123.88
> Microsoft .NET Framework
> Version 4.7.03190
> 
> Installed Version: Enterprise
> 
> Visual C++ 2019   00433-90000-00004-AA359
> Microsoft Visual C++ 2019
> 
> ASP.NET and Web Tools 2019   16.2.290.48383
> ASP.NET and Web Tools 2019
> 
> C# Tools   3.2.0-beta4-19359-03+15b43b33901c88f68ef43f8314b5a2457716780d
> C# components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.
> 
> GitHub.VisualStudio   2.8.0.6762
> A Visual Studio Extension that brings the GitHub Flow into Visual Studio.
> 
> IntelliCode Extension   1.0
> IntelliCode Visual Studio Extension Detailed Info
> 
> Microsoft JVM Debugger   1.0
> Provides support for connecting the Visual Studio debugger to JDWP compatible Java Virtual Machines
> 
> Microsoft MI-Based Debugger   1.0
> Provides support for connecting Visual Studio to MI compatible debuggers
> 
> Microsoft Visual C++ Wizards   1.0
> Microsoft Visual C++ Wizards
> 
> Microsoft Visual Studio VC Package   1.0
> Microsoft Visual Studio VC Package
> 
> NuGet Package Manager   5.2.0
> NuGet Package Manager in Visual Studio. For more information about NuGet, visit https://docs.nuget.org/
> 
> ProjectServicesPackage Extension   1.0
> ProjectServicesPackage Visual Studio Extension Detailed Info
> 
> Test Adapter for Boost.Test   1.0
> Enables Visual Studio's testing tools with unit tests written for Boost.Test.  The use terms and Third Party Notices are available in the extension installation directory.
> 
> Test Adapter for Google Test   1.0
> Enables Visual Studio's testing tools with unit tests written for Google Test.  The use terms and Third Party Notices are available in the extension installation directory.
> 
> Visual Basic Tools   3.2.0-beta4-19359-03+15b43b33901c88f68ef43f8314b5a2457716780d
> Visual Basic components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.
> 
> Visual Studio Code Debug Adapter Host Package   1.0
> Interop layer for hosting Visual Studio Code debug adapters in Visual Studio
> 
> Visual Studio Tools for CMake   1.0
> Visual Studio Tools for CMake
> 
> Visual Studio Tools for CMake   1.0
> Visual Studio Tools for CMake

2. Download deps:
[64 bit, Release mode only](https://bintray.com/vojtechkral/Slic3r-PE/download_file?file_path=destdir-64.7z)
3. Unzip the deps to C:\local
4. Clone the git repo into MVS and open the command line from: tools/command line/DCP
5. Building the dependencies package
cd deps
mkdir build
cd build
cmake .. -G "Visual Studio 16 2019" -DDESTDIR="C:\local\destdir-custom"
msbuild /m ALL_BUILD.vcxproj
6. goto home directory the slicer and make it:
mkdir build
cd build
cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="C:\local\destdir-64\usr\local"
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.17763.
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: C:\local\destdir-64\usr\local (from cache or command line)
Building without Win10 Netfabb STL fixing service support
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Boost (missing: system filesystem thread log locale regex)
  (found version "1.63.0")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindBoost.cmake:2142 (find_package_handle_standard_args)
  CMakeLists.txt:237 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/user/source/repos/prusa3d/PrusaSlicer/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/user/source/repos/prusa3d/PrusaSlicer/build/CMakeFiles/CMakeError.log".

I can not make the slicer use MVS 16 2019 and cmake. I can see that "Building without Win10 Netfabb STL fixing service support" also.
I want make the slicer from sources and i want make the with Win10 Netfabb STL fixing service support.

vojtechkral, 2019-07-30:
The dependencies bundles were generated for MSVS 2013, they are no longer usable. Sorry. You will need to build dependencies yourself. See #2688 - it's the same issue. Closing this one.

closed 2019-07-30 by vojtechkral

jayraaj, 2019-08-04:
Can I have the all the dependencies with an exact version so I can build them on VS 2017?

vojtechkral, 2019-08-04:
@jayraaj Yes, the dependency build should work on VS 2017.

jayraaj, 2019-08-04:
Means which build? I want to build on Windows.

jayraaj, 2019-08-04:
Could you please shy the link

vojtechkral, 2019-08-04:
@jayraaj Yes, the details are in the [how to document](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md). It's somewhat outdated but the dependency build part is still accurate provided you use the right CMake generator.

jayraaj, 2019-08-07:
I want to build all the dependencies on my end with VS 2017 so could you please let me know all dependencies and its version for PRUSA 2.0?

vojtechkral, 2019-08-07:
@jayraaj The list of dependencies along with versions and build flags for MS Windows specifically is in the `deps/deps-windows.cmake` file.