prusa3d/PrusaSlicer #2697: [FEATURE REQUEST] Print the first layer skirt with every extruder

Schild0r, 2019-07-30:
As the title says I think it would be great to have an option for the skirt in the first layer to be printed by every extruder (resulting in [nozzle count] times the amount of set skirt lines for the first layer).

This might help multi nozzle prints in several ways:
1. the ooze shield would sit on a wider base which would effectively serve as a brim for it which is prone to detaching from the build plate due to its small bottom surface area
2. it would make sure that at the start of a print every nozzle gets primed
3. it would cycle every tool through a tool change which results in the retract_length_toolchange being applied to every extruder at the very start of the print (which is currently missing)
3a. this could maybe lead to the standby_temperature_delta not being necessary anymore if the retract_length_toolchange is longer than the nozzle+heatbreak

bubnikv, 2019-07-31:
This should already work if you:
1) Have multiple extruders defined
2) You have "single extruder multi material" disabled
3) You defined enough of skirt lines to be extruded.

Schild0r, 2019-07-31:
Well it does not look like it unfortunately. In fact the first layer only will have half the skirt lines when using multiple nozzles (I think there is something misimplemented that prevents the second nozzle from printing the other half).

The part:
[image]

First layer:
[image]

Second Layer:
[image]

I also tried enabling "Prime all printing extruders" although this should only have an effect on single nozzle multi material printers. As expected it didn't change anything.

The "only half the number of skirt lines" issue has been a problem for at least two years now apparently (thread #469) or maybe it is just in my special case (which I actually don't consider that special tbh). I will attach my profiles, So if there are contradictions you can maybe spot them
[Profiles.zip](https://github.com/prusa3d/PrusaSlicer/files/3452793/Profiles.zip)

Schild0r, 2019-08-01:
Another problem: the unused nozzle won't undo the retract_length_toolchange at the end of a print.
Seems a little bit like multi nozzle setups are neglected to me. This could elegantly be fixed with contidional gcode (which is possible in Slic3r)

closed 2019-08-02 by Schild0r