prusa3d/PrusaSlicer #2698: Problems with autogenerated gcode

Zalama, 2019-07-31:
### Version
2.0.0

### Operating system type + version
Win 10

### Behavior
Slicer autogenerates M109 only AFTER start g-code. It should be performed before.

Also after printing slicer doesn't command bed and hot end to cool down with commands like:
M104 S0;
M140 S0;

I had to add those manually to end G-code

bubnikv, 2019-08-09:
Maybe you don't want the bed to be switched off automatically, so your print will not pop off?
IMHO we don't need to change the behavior, but to document the current behavior.

Sono-Gomorrha, 2020-12-08:
I found this irritating as well. In the 2.2.0 release I modified the custom end GCode under printer settings and added M140 S0 ;Turn-off bed

rtyr, 2021-02-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/644. Closing.

closed 2021-02-03 by rtyr