prusa3d/PrusaSlicer #2699: Start G-code issue

vdauer, 2019-07-31:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

vdauer, 2019-07-31:
. Prusa Slicer version 2.0 
. Windows 10 64bit
.Lulzbot Taz6 with Areostruder tool head firmware is Marlin

. I copied and pasted the start g-code from Lulzbot and pasted it in the printer settings custom gcode section. 

[image]

In the start gcode placeholders "{}" are used for bed and extruder temperatures. When I try to slice the model I get this error. 

[image]

 In Cura and Simplified 3D the same start gcode is used and the slicer program pulls the temperature and puts them in the final source gcode. And it looks like this.

[image]

If I replace for example M140 S{material_bed_temperature} with M140 S110, the output is good. But this is a step I'd like to avoid.

Thanks so much.

BerndJM, 2019-07-31:
"material_bed_temperature" didn't exist.
Availabel variables are "bed_temperature" or "first_layer_bed_temperature"

Schild0r, 2019-07-31:
You can only set variables that refer to actual settings... The name of the variable for each setting will appear in the tooltip as you hover your mouse over it

vdauer, 2019-08-02:
This worked 
M104 S[first_layer_temperature] ; set extruder final temp
M109 S[first_layer_temperature] ; wait for extruder final temp
M190 S[first_layer_bed_temperature] ; wait for bed final temp

The softening temps I put in directly.
Thanks for the help guys!

closed 2019-08-02 by vdauer