prusa3d/PrusaSlicer #2700: Smoothing the upper layers

artdezigner, 2019-07-31:
Good afternoon, Your slicer really works better than Cure, but I really miss the function of smoothing the upper layers, please add this feature and then I can completely switch to PrusaSlicer.

bubnikv, 2019-08-01:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2700
Closing this one.

closed 2019-08-01 by bubnikv

artdezigner, 2019-08-01:
where's the answer to the question?

tkircher, 2019-08-02:
They meant #458 
The answer is that it's not a priority, please search the issues database before opening a new issue. ;)

artdezigner, 2019-08-02:
I read, sadly, that you can not do the desired function for several years. Also with other useful functions, usb, languages... all this makes your slicer defective for normal operation.

tkircher, 2019-08-02:
If you're volunteering to add Russian support, definitely let the developers know.

artdezigner, 2019-08-02:
let's try. How to do this? What about printing via usb? This is also a very necessary function, as well as smoothing.

tkircher, 2019-08-02:
Literally nobody uses smoothing. As for printing via USB:

https://blog.prusaprinters.org/prusaslicer-2-release/#11

artdezigner, 2019-08-02:
how not to use smoothing? You have given a link where you are asked to do this feature for as long as two years. Where people write that it is necessary and I continue to use Cura

rongith, 2019-08-06:
a lot of people don't do because the feature is not implemented. but how can you say that nobody uses smoothing? and there is no relation to printing through usb. i do smoothing every day with a fork from supermerill.