prusa3d/PrusaSlicer #2701: Feature Request: Console output options

bobmonkey07, 2019-07-31:
### Version
Version 2.0.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
MK3s

_Is this a new feature request?_

Is it possible to add details such as (print time) or (used filament) to the file output, or is that GUI only?

BerndJM, 2019-07-31:
Isn't it already present at the end of the gcode file? 
For me a typical file looks at this position like this:

> ; filament used [mm] = 2862.6
> ; filament used [cm3] = 18.3
> ; filament used [g] = 22.6
> ; filament cost = 0.5
> ; total filament used [g] = 22.6
> ; total filament cost = 0.5
> ; estimated printing time (normal mode) = 1h 48m 42s

bobmonkey07, 2019-08-01:
Huh. I'd skimmed a few files and missed that, Thanks! It would be nice to have it easily output with the file name like it has with the whole GUI.

bobmonkey07, 2019-08-01:
After looking at it more, console generated files don't seem to generate ;total filament used [g] or cost, but the other fields are fine.

vojtechkral, 2019-08-02:
@bobmonkey07 If I understand right, the problem is that when slicing from CLI, the filament / time stats are not written in the gcode? If that's the case, could you please update the issue description and title?

bobmonkey07, 2019-08-02:
The filament in (g) is not, but time is. My issue is I'd like to append those on the file name output from console like I can from the GUI.

closed 2026-09-29 by prusaslicer-bot[bot]