prusa3d/PrusaSlicer #2702: [FEATURE REQUEST] Add an option for "convex raft"

RossKor, 2019-07-31:
### Version
2.0.0

### Feature Requested
Add option to create convex rafts (for example - have the first layer of the raft fill the entirety of the skirt).

### Justification
Some materials are prone to curling - when using a raft to try to achieve a successful print the printhead may trip on the raft itself due to curling.

This can be prevented if the raft had no internal voids/gaps for the printhead to trip on.

bubnikv, 2019-08-09:
Would you please provide pictures?

RossKor, 2020-01-04:
[image]

bubnikv, 2020-01-06:
Why?

bubnikv, 2021-02-22:
Actually most of our users want the opposite: Generate rather no raft at all.

closed 2021-02-22 by bubnikv