prusa3d/PrusaSlicer #2703: Feature Request: USB Support

rdragonrydr, 2019-08-01:
Based on what I've heard from the documentation I've read so far, PrusaSlicer does not have the ability to directly control the printer; instead, the g-code is written to an SD card and read from there. I don't have a RPI0W (would a RPI3 work?), either.

To round out the features, would it become feasible to add USB printing support, or the ability to connect to the printer via USB to manage the files on the SD card (create/update/delete)?
It can already connect to update firmware, though I'm guessing that has to be via a different method altogether.

Pronterface does not really work properly on modern Linux, and it's not supposed to have support for a few of the newer features you've added.

federico-galli, 2019-08-01:
Repetier-Host works on linux with slic3r/prusaslicer

vojtechkral, 2019-08-01:
> would a RPI3 work?

Yes.

In any case, this is a duplicate of #2046

closed 2019-08-01 by vojtechkral

bubnikv, 2019-08-01:
As of now, the management considers this feature to be a low priority for
our customers.

rdragonrydr, 2019-08-01:
> > would a RPI3 work?
> 
> Yes.
> 
> In any case, this is a duplicate of #2046

Oops. I had tried looking for similar issues, but missed that one somehow.
Is there anything special needed for the automatic pause and power-panic to work with Octoprint, though, if it's an RPI3 over USB instead of a zero over GPIO?

vojtechkral, 2019-08-02:
> Is there anything special needed for the automatic pause and power-panic to work with Octoprint, though, if it's an RPI3 over USB instead of a zero over GPIO?

I'm not sure what you mean. There's shouldn't be any difference between RP3 and RP0 in terms of GPIO or USB. Power panic is, as far as I know, _not_ supported by Octoprint regardless of RP.

rdragonrydr, 2019-08-02:
I may have been wrong on that. It's been a while now, so I don't recall what exact feature, but there was supposed to be something that Prusa was working with Octoprint to add. It was some kind of automatic pause feature (probably on running out of filament), I think?