prusa3d/PrusaSlicer #2704: Feature Request: Non-planar slicing (slightly curved slicing)

probonopd, 2019-08-01:
A recent research paper is describing "slightly curved slicing for 3-axis printers", also known as "non-planar slicing".

[image]

Photo credit: https://hackaday.com/2016/07/27/3d-printering-non-planar-layer-fdm/

> CurviSlicer: Slightly curved slicing for 3-axis printers
> 
> Abstract : Most additive manufacturing processes fabricate objects by stacking planar layers of solidified material. As a result, produced parts exhibit a so-called staircase effect, which results from sampling slanted surfaces with parallel planes. Using thinner slices reduces this effect, but it always remains visible where layers almost align with the input surfaces. In this research we exploit the ability of some additive manufacturing processes to deposit material slightly out of plane to dramatically reduce these artifacts. We focus in particular on the widespread Fused Filament Fabrication (FFF) technology, since most printers in this category can deposit along slightly curved paths, under deposition slope and thickness constraints. Our algorithm curves the layers, making them either follow the natural slope of the input surface or on the contrary, make them intersect the surfaces at a steeper angle thereby improving the sampling quality. Rather than directly computing curved layers, our algorithm optimizes for a deformation of the model which is then sliced with a standard planar approach. We demonstrate that this approach enables us to encode all fabrication constraints , including the guarantee of generating collision-free toolpaths, in a convex optimization that can be solved using a QP solver. We produce a variety of models and compare print quality between curved deposition and planar slicing.

Source:
https://hal.archives-ouvertes.fr/hal-02120033

Further reference:
* https://hackaday.com/2016/07/27/3d-printering-non-planar-layer-fdm/

vojtechkral, 2019-08-01:
I had a look at this some time ago, but couldn't get their perl script to output useful results (I don't remember what the problem was though). Did you manage to replicate their results? Do you have a gcode you could share, preferably for a Prusa printer?

probonopd, 2019-08-01:
[wing.zip](https://github.com/prusa3d/PrusaSlicer/files/3458628/wing.zip)

I created this using the instructions from https://github.com/makertum/non-planar-layer-fdm/tree/master/examples using Slic3rPE-1.41.1+linux64-full-201810261150.AppImage.

It is being printed on the MK2S right now.

vojtechkral, 2019-08-03:
Hm. So far I had little luck printing this, it comes out very underextruded. I might try again next week with a different printer.

In any case, while this is certainly interesting, the scope of usefulness seemskind of limited...

davidelang, 2019-08-03:
a quadrature encoder requires two sensorts 50% out of phase with each other. 
They may be in one houseing, which would make the alignment less critical

David Lang

ricorx7, 2019-08-22:
An updated example using Slic3r 

https://tams.informatik.uni-hamburg.de/research/3d-printing/nonplanar_printing/index.php

https://github.com/Zip-o-mat/Slic3r/tree/nonplanar

https://youtu.be/km1lvuva5mI

lloyb, 2019-09-06:
The future of 3D printing right here, get on it Prusa and E3D

espr14, 2019-10-01:
Easier possibility could be printing in predefined curved layers instead of flat horizontal planes. Such layer could be sinus or defined by STL. This would improve strength of the result. Since the shape of curved layer is fixed, it is easy to implement. Even easier could be implementation of tilted flat layers. 

[image]

fiveangle, 2019-10-01:
Benchy’s of the world, prepare to have your cabin roof severely upgraded 😛

mb3m3r, 2020-09-07:
Hi, im having some trouble achieving this non planar printing.

I'm using windows 10 and followed all the setps from makertum (https://github.com/makertum/non-planar-layer-fdm)
1) Installed strawberry perl (when I run the CPAN client and try to install Math:Round i get "Database was generated on Mon 07 Sep 2020 Math: :Round is up to date (0.07)."
[image]
2) Downloaded the scrip and added the absolute path of the script in the output options.
[image]
3) Checked Use relative E distances and Verbose Gcode and made sure Vase mode was UNCHECKED.
[image]
[image]
4) Loaded the Slicer setting bundle (and re-write the path for the script)
[image]

But the preview for CUBE2 is still just a cube, no waves on the top layers. I exported the Gcode and viewed it on Repetier Host and it's the same, no waves. 
[image]

I'm not sure how to "Make sure the script file is executable by chmod 755 non-planar-layer-fdm.pl". I tried running that command on the CMD but I get "CHMOD is not recognized as a command" and when I go to the folder containing the script and run it it just goes to the next line of the CMD waiting for input or something but no errors, so I guess its good? I do get a warning for opening the scrip  (I always click open) when exporting the gcode in slic3r. 
[image]
[image]

Am I doning something wrong or missing something?

fiveangle, 2020-10-02:
https://github.com/makertum/non-planar-layer-fdm/issues/new/choose

probonopd, 2022-09-17:
Wouldn't it nice if one could print nonplanar curved top surfaces? These are often used in industrial design, e.g., here:

[image]

I imagine that most of the layers would be printed normally, but then a few nonplanar ones on top, and finally nonplanar ironing. Something along these lines (section view of the layers):

[image]

DragonFinlay, 2022-09-17:
I'm confident that this is something we'll see a lot in the (hopefully) near future. Currently, I feel like there are still big steps to be made here in terms of print head design, safety, preventing crashes, and getting reasonable quality.

About ironing: Since it pretty much relies on the flat part of the nozzle being parallel to the topmost surface, I'm not sure if a clean result is possible when essentially "burying" the edge of the nozzle into the lines.

probonopd, 2022-09-17:
> About ironing: Since it pretty much relies on the flat part of the nozzle being parallel to the topmost surface, I'm not sure if a clean result is possible when essentially "burying" the edge of the nozzle into the lines.

Damn. Right. That must be why some people are using tiltable hotends. But that requires new 3D printers or challenging upgrades. So no ironing then.

cfontes, 2023-03-23:
Bump for this.

One of the biggest "pending" features in 3D printing. Surface quality would be so improved, worst then "The Lines" is "The Stairs", Lines are bearable but the Stairs generally kill the surface finish, and for printing with largers nozzles it's almost impossible to avoid.

Hopefully it sees some love this year, propbably hard to implement but definetely worth it, this and textured modifiers ( Idea maker already has it ) are the 2 most important features at least to me. surface quality is so important to present 3D printing to other people and this makes it a lot better.

Goog luck.

EiNSTeiN-, 2023-07-02:
I ported the old nonplanar fork that only works with a 5-years old version of slic3r to PursaSlicer 1.6.0, here it is in case anyone wants to give it a try: https://github.com/EiNSTeiN-/PrusaSlicer.

As far as my testing went, it works just as the original did, which unfortunately doesn't seem to produce results for real life models. I intend to put some time towards understanding why it doesn't work for models like the 3DBenchy. There are lots of other issues, like being super slow and not being able to run in parallel. As it is, the code likely will never be merged upstream, it would need a lot of work for that to happen.

Hopefully this is one step in the right direction, enjoy!

[image]

probonopd, 2023-07-26:
Nice video by E3D on Non-Planar 3D Printing:
https://www.youtube.com/watch?v=Kym0sJMRk7s

Felix14-v2, 2026-09-18:
I think that feature requests shouldn't be closed.

mssalvatore, 2026-11-08:
I think this would be a truly amazing feature.