prusa3d/PrusaSlicer #2706: "External perimeters" speed setting applied to internal when "External perimeters first"

thehans, 2019-08-01:
### Version
2.0.0+linux64

### Operating system type + version
Linux Mint 19

### 3D printer brand / version + firmware version (if known)
custom reprap
Marlin 1.1.9

### Behavior
I noticed my print going slowest on the innermost perimeters.  The "Speed" -> "External perimeters" is set to 50%, and "Layers and Perimeters" -> "External perimeters first"

Also I was going to try to verify/show the speed differences based on the layer preview speed setting, but the scale is so far off it (max at 1500mm/s) that there is no discernable color difference across the entire print.  It would be nice if the maximum of the speed scale was set to the maximum speed of any moves on the given model (or the toolpath currently within view).
[image] 
I can open this as a separate issue if wanted.

Jebtrix, 2019-08-02:
I cannot duplicate the perimeter speed swap behavior. Maybe post the project file if its possible. 

The maximum speed value in "Speed View Filter" is usually your Travel speed. If you set Travel to **1** then you will see the speed scale with actual print moves speed range.

thehans, 2019-08-03:
I tried setting the travel speed to 1 (Print Settings -> Speed -> Speed for non-print moves -> Travel Speed), and it scaled the *minimum* of the Speed View color scale to 1.0, but maximum remains 1500mm/s.  I don't see a setting that high anywhere in my configuration.

thehans, 2019-08-03:
Well now that I can easily verify the actual speeds visually, I can't reproduce the issue.  I think I must have mistaken the transition to "Small perimeters" speed as a backwards "External perimeter" slowdown.

closed 2019-08-03 by thehans