prusa3d/PrusaSlicer #2707: Extensive syslog logging on ubuntu, prusaslicer not responding

richiewebgate, 2019-08-02:
### Version
Version 2.0.0+linux64
PrusaSlicer-2.0.0+linux64-201905201652.AppImage

### Operating system type + version
Ubuntu 19.04

### 3D printer brand / version + firmware version (if known)
MK3S, but not relevant to this issue

### Behavior
When slicing an object, rsyslogd consumes extensive cpu resources, renders prusaslicer app inresponsive for several minutes and bloats the system log. 

*Steps to reproduce:*

Start App, import any STL Object, click "Slice now".
Program shows "Slicing...", window turns "grey" for minutes (like not responding) and /var/log/syslog shows "millions" of lines like:

Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005f9
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005f5
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005f9
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fd
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005ff
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fb
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fb
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005ff
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fc
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005f8
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005f8
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fc
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fe
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fa
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fa
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fe
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fd
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005f9
Aug  2 10:08:32 localhost org.gnome.Nautilus[2451]: nouveau: #0110x000005fd
...repeated approx. a million times :-)

I am having the impression, that there is some unnecessary logging enabled which causes the program to become unresponsive due to rsyslogd overload?

Result: PrusaSlicer not suited for production use on ubuntu. Have to start a windows machine again.

bubnikv, 2019-08-02:
Intersting, if it was unusable, I would expect this issue to be reported
very often.

I have no idea what this logs mean. PrusaSlicer only logs to console, and
only if the logging is enabled, which normally is not (you may use the
--loglevel=number command line switch).

richiewebgate, 2019-08-02:
I am using the AppImage version.. so I am not sure if this also accepts command line arguments. 
I just tried starting the AppImage with loglevel arguments, but no difference. btw when I start the AppImage from commandline, I get all the "nouveau: xxx" loggings right in the console until I hit CTRL-C. Just noticed that this happens sometimes even right after loading an STL, even before clicking the slice button.
....
nouveau: 	0x0004792c
nouveau: 	0x00000001
nouveau: 	0x00107b00
nouveau: 	0x00000000
nouveau: 	0x00215000
nouveau: 	0x0000004b
nouveau: 	0x1000f010

(prusa-slicer:25413): GLib-GObject-WARNING **: 13:53:11.184: g_object_weak_unref: couldn't find weak ref 0x7f87cf947d30(0x47d0ec8)

(prusa-slicer:25413): GLib-GObject-WARNING **: 13:53:11.211: g_object_weak_unref: couldn't find weak ref 0x7f87cf947d30(0x3c87ac8)

(prusa-slicer:25413): GLib-GObject-WARNING **: 13:53:11.226: g_object_weak_unref: couldn't find weak ref 0x7f87cf947d30(0x3e848a8)

(prusa-slicer:25413): GLib-GObject-WARNING **: 13:53:11.270: g_object_weak_unref: couldn't find weak ref 0x7f87cf947d30(0x36bc098)

** (prusa-slicer:25413): CRITICAL **: 13:53:11.287: gail_notebook_real_remove_gtk: assertion 'obj' failed

** (prusa-slicer:25413): CRITICAL **: 13:53:11.291: gail_notebook_real_remove_gtk: assertion 'obj' failed

I will try to build and install without AppImage and see if that helps

bubnikv, 2019-08-02:
>   I am using the AppImage version.. so I am not sure if this also accepts
command line arguments.

it does

bubnikv, 2019-08-02:
https://nouveau.freedesktop.org/wiki/
https://wiki.archlinux.org/index.php/Nouveau
https://nouveau.freedesktop.org/wiki/TroubleShooting/

It looks like the OpenGL driver is leaking these log messages.

richiewebgate, 2019-08-02:
ok thanks.. so I just found out that the .tar.bz2 download option doesn't need to be compiled, great!
With that one, same "nouveau" errors. So I tried it on a second machine, also with ubuntu 19.04 but a different GPU, and there it looks like it's working fine without excessive logging and also no gray "not responding" phases. 

.. so if I am the only one, then just consider it solved, I will copy the STL output from my 3D Desktop to my notebook for processing in the prusaslicer. but would have been cool to do all the steps on just one computer instead of two.

closed 2019-08-02 by richiewebgate

richiewebgate, 2019-11-05:
after upgrading to ubuntu 19.10, I had to switch from "nouveau" opensource gfx drivers to nvidia proprietary drivers due to general stability issues with nouveau. Now prusa slicer works as it should.