prusa3d/PrusaSlicer #2708: Unable to remove support material/can't change support "inflation distance"

cadowd, 2019-08-02:
### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s

### Behavior
When slicing support material is placed inside holes where it cannot be removed (if the hole is longer than a normal drill bit). This is not the case when using simplify3d, which doesn't place the support material along the sides of the part. Even when using support blockers inside the hole, this material is still there. There is also no option to reduce the distance that the support extends outside the part boundaries (not the minimum distance of the support to the part), similar to the extra inflation distance in simplfy3d. This would help to avoid this issue.

As it stands, this support material can't be removed and blocks a mating part, so I've had to use simplify3d to slice this part. It would be great it there were some options to avoid this problem in the future in prusaslicer.

[image](https://user-images.githubusercontent.com/23720790/62377525-39e11780-b543-11e9-95ae-0a408774ed4a.png)

#### Project File (.3MF) where problem occurs

[hinge.zip](https://github.com/prusa3d/PrusaSlicer/files/3461740/hinge.zip)

bubnikv, 2021-03-22:
Testing snug supports

[image]

bubnikv, 2021-03-23:
Fixed with af9c7c967f08fcf22ffa628c324c90eda65230f1

    Implementing a new switch for the shape of support towers:
    expanded to a grid (the old way) vs.
    snug (like the upstream Slic3r, Cura or Ideamaker).
    
    Snug supports suffered from the degeneracies when merging overhang islands
    over a large number of layers when projecting the support towers down.
    We borrowed the idea & a bit of code from Cura by simplifying the support
    polygons by closing the concave cracks, see the smooth_outward() function
    and the MutablePolygon class.

closed 2021-03-23 by bubnikv