prusa3d/PrusaSlicer #2709: Extra gcode line at start move nozzle at defined Lift Z height?

weirdoku, 2019-08-02:
**Version**
PrusaSlicer-2.0.0+win64-201905201652

**Operating system type + version**
Windows 10

**3D printer brand / version + firmware version (if known)**
Ender 3

**Behavior**

Here's the section of gcode I'm referring to:

; Ender 3 Custom Start G-code
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
;G4 S30 ; Dwell 30 Seconds for bed to reach temp
M104 S200 ; set extruder temp
M109 S200 ; wait for extruder temp
G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.2 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X2.0 Y200.0 Z0.2 F5000.0 ; Move to side a little
G1 X2.4 Y20 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 E-2.50000 F3300.00000
G92 E0
### G1 Z0.600 F6000.000
G1 X71.558 Y63.208
G1 Z0.200
G1 E2.50000 F3300.00000
G1 F2400
G1 X72.813 Y62.025 E2.55837
G1 X74.156 Y61.045 E2.61467
and so on...

Everything is fine up until the bolded out line "G1 Z0.600 F6000.000". For some reason Prusa Slicer adds in this line to move the nozzle to the "Lift-z" height, then moves to the starting XY position, then down again to the "First Layer Height" then starts printing.

It seems redundant to have that line of gcode to move the nozzle up to the Lift-z height but only to go down again to the First layer height. Or am I missing something...? I'm not using a Prusa machine so I don't know what they do at the start, both Prusa Slicer and S3D has this line added and only Cura doesn't have this extra line, it moves to the XYZ start with a gcode line like so **G0 F3600 X106.651 Y103.697 Z0.2**.

Is it reaaaally causing an issue... no but it adds some unnecessary z movements where it doesn't really need to. Please correct me if I'm wrong or educate me if it serves a some purpose that I just can't think of!

Thank you!

AndyWo83, 2019-08-05:
I have the same problem too.
I also have almost the same start code as @weirdoku.
After the line I move up to Z1.0 and then he should actually drive to the starting point of the print object, but he moves the print head to the Z-Lift position or first layer height Z0.2 and then to the starting point of the print object.

bubnikv, 2019-08-09:
Would you please post your config?
Do you have Z-hop enabled?
IMHO this is not an issue at all. Please consider whether opening such an issue is meaningful to the project, as currently there are 858 issues open, with many much more important than this IMHO non-issue.

AndyWo83, 2019-08-12:
I think in my case i have a other problem.
Z-Axis is moving first bevor X and Y in position.

See after the Filament gcode:
; Filament gcode
M900 K0
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 E-1.20000 F2100.00000
G92 E0
G1 Z0.200 F8400.000        <----- 2nd
G1 X124.125 Y80.016        <----- 1st
G1 E1.20000 F2100.00000
G1 F1200
G1 X125.542 Y78.219 E1.26888
G1 X126.651 Y77.105 E1.31620
G1 X127.760 Y76.174 E1.35979 

My opinion is, that X and Y have to move to the first position, and then Z has to move down.

I use these two prime lines of cura and every time after the end of the line, the axis moves back down and pulls the line with me, when the x and y position are approached.

lukasmatena, 2019-08-13:
Would you please post your config like @bubnikv asked you to?

weirdoku, 2019-08-13:
> Would you please post your config?
> Do you have Z-hop enabled?
> IMHO this is not an issue at all. Please consider whether opening such an issue is meaningful to the project, as currently there are 858 issues open, with many much more important than this IMHO non-issue.

[config.zip](https://github.com/prusa3d/PrusaSlicer/files/3498008/config.zip)

That's my entire config bundle.

What I'm trying to explain is the slicer adds an extra line of gcode to move the nozzle up/down to the defined z-lift height, then moves to XY start postion, then down again to the first layer height then starts printing.

; Ender 3 Custom Start G-code
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
;G4 S30 ; Dwell 30 Seconds for bed to reach temp
M104 S200 ; set extruder temp
M109 S200 ; wait for extruder temp
G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.2 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X2.0 Y200.0 Z0.2 F5000.0 ; Move to side a little
G1 X2.4 Y20 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 E-2.50000 F3300.00000
G92 E0
**G1 Z0.600 F6000.000** - move up/down to z-lift height
**G1 X71.558 Y63.208** - move to XY start position
**G1 Z0.200** - move down again to first layer height
G1 E2.50000 F3300.00000 - starts printing
G1 F2400
G1 X72.813 Y62.025 E2.55837
G1 X74.156 Y61.045 E2.61467

After the gcode line:

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

It moves the nozzle down to the z-lift height with:

G1 Z0.600 F6000.000 (this line gets added in by default)

Which basically cancels out G1 Z2.0 F3000

Like Andy says, the three bolded out lines above could be combined to the line below and remove G1 Z0.600 F6000.000 entirely?

G1 Z0.200 X71.558 Y63.208 F6000

I think it makes a cleaner travel move before printing.

bubnikv, 2019-08-19:
You have your Z-lift on retract enabled in your config
retract_lift = 0.4

Therefore slicer just follows what it has been told: Z hop on every travel.

By the way, Slicer does not analyze your start G-code to the extent to understand, whether your start G-code extruded anything (for example some extruder priming lines), so it always Z hops.

weirdoku, 2019-08-19:
> You have your Z-lift on retract enabled in your config
> retract_lift = 0.4
>
> Therefore slicer just follows what it has been told: Z hop on every travel.
>
> By the way, Slicer does not analyze your start [...]

I'm not sure if you're understanding what I'm trying to explain... I already mentioned I have z-lift enabled in my original post. The slicer is adding an unnecessary line of gcode that creates an unnecessary nozzle move, which like myself and Andy have also said, that line of gcode could be removed or combined to create a cleaner travel move to the start of a print.

I will close this 'issue' and manually edit my gcode.

closed 2019-08-19 by weirdoku