prusa3d/PrusaSlicer #2710: Multi nozzle / toolchange retraction issues

Schild0r, 2019-08-02:
So recently decided to use my second nozzle for the first time but I ran into some issues on the slicing side.
Although I really like PrusaSlicer it's got some issues regarding multi nozzle setups.
So I wanted to dedicate a single thread to these issues I found so that they don't get lost in my feature request and the mass of feature request the team is surely bombarded with.

1. If you want to slice a model in which the second extruder won't be used in the first layer, the amount of skirt lines in the first layer is only half of the specified amount (duplicate from #469)

2. The unused nozzle is oozing onto the print until the first toolchange triggers the "toolchange retraction" specified in the print settings under multiple extruders
Possible fixes:

2a) For the first layer the order of printing is: 
   1. Extruder_0: Skirt
   2. Extruder_0: Part
   3. Extruder_1: Skirt
   4. Extruder_1: Part
But to prevent the Second extruder from oozing onto the print it should rather be
   1. Extruder_0: Skirt
   2. Extruder_1: Skirt
   3. Extruder_0: Part
   4. Extruder_1: Part
So the toolchange would be triggered before anything of the part is printed

2b) Introduce the placeholders [first_used_extruder] and [last_used_extruder] as well as [{first/last}_unused_extruder_{0-n}] (with 0 to n-1 beeing the amount of nozzles for printers with more than 2 nozzles)
for the custom start and end gcode so that it can be taken care of in the custom gcode

2c) at the start of the print when the nozzles are hot perform a retraction with the length of [retract_length_toolchange] on both extruders and then undo it on the first used extruder only

3. Temperatures: If you let PrusaSlicer handle the heat up of the nozzles it will heat them up to [first_layer_temperature] + [standby_temperature_delta] which is correct but then it will heat up the first used nozzle to [temperature] where it should be [first_layer_temperature] instead

4. At the end of the print the last unused extruder won't undo the toolchange retraction always causing an unprimed nozzle on the next print
Possible fixes: see 2b) or simply undo it by default at the end of multi nozzle prints

giPeteR, 2020-02-26:
**Toolchange retraction doesn't work for 1st layer.**
I have the same problem (wrote it in a different post before finding this excellent description).

OS: Win 10.
FW: Marlin with two direct drive extruders.
PS: 2.2.0-alpha2

Purpose:
1. Prime T0 & T1.
2. Retract either both, or only the one not being used the 1st layer, to prevent ozing.
3. Print 1st layer with T0.
4. Retract T0 (tool change retraction).
5. Unretract T1  (tool change un-retraction).
6. Print 2nd layer with T1 (or both T0 + T1).

Problem is that step 2 & 5 is missing.
I tried to manually retract them before start, but regardless using absolute or relative extrusion, Slicer always resets the distance after initial G-code, and any pre-retraction gets lost (i.e. no corresponding unretract).
T0 & T1 order doesn't matter. Same or different filament presets doesn't matter.

So before 1st layer is the normal layer change (un)retract, but no tool change (un)retract:

> ; printing object Retract test 1A.stl id:0 copy 0
G1 E-0.60000 F1800.00000 ; retract
G1 X114.225 Y65.225 F7800.000 ; move to first perimeter point
G1 Z0.500 ; restore layer Z
G1 E0.80000 F1800.00000 ; unretract

After 1st layer and when swapping tools, the actual tool change retract occurs.
The one I'd like to see before 1st layer as well:

> ; stop printing object Retract test 1A.stl id:0 copy 0
G1 E-5.00000 F1800.00000 ; retract for toolchange

...and then it works for the rest of the layers.

Is there some work around with 
---
**Orphan filament End G-code**
One odd thing, why is all filament custom ending G-codes showing at the end of the print?
Even if that filament has not been used and has no Start G-code?
Example:

> ...
> T1
> Filament 1 Custom Start G-code
> Filament 1 Custom End G-code
> T0
> Filament 0 Custom Start G-code
> Filament 0 Custom End G-code
> Filament 1 Custom End G-code -- NO corresponding Start G-code. Should not be here, or?
> End of print.
> 

---
The G-code:
[Retract test 1A.zip](https://github.com/prusa3d/PrusaSlicer/files/4252242/Retract.test.1A.zip)

giPeteR, 2020-02-26:
This is more or less the same...
[How to call toolchange_gcode in the start gcode? #2720](https://github.com/prusa3d/PrusaSlicer/issues/2720#issue-476565099)

giPeteR, 2020-03-21:
I made a post processing code in Python that solves this.
https://github.com/giPeteR/PrusaSlicer-DualExtruder

closed 2026-09-29 by prusaslicer-bot[bot]