prusa3d/PrusaSlicer #2712: PrusaSlic3r adds unwanted geometry on top of last layer

Torsten85, 2019-08-03:
### Version
2.0.0+
### Operating system type + version
MacOS 10.14

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
The Slic3r adds unwanted geometry on the last layer. In "3d editor view" this geometry is not there. In "Preview" and also in the printed object, you can clearly see it.

[image]

#### Project File (.3MF) where problem occurs
[Project.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3463566/Project.3mf.zip)

neophyl, 2019-08-03:
its is there, the top of your model is not perfectly level when compared to the bottom of your model so those areas are just high enough for slicer to register them as another layer.  Basically theres a very very slight slope

bubnikv, 2019-08-06:
See
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)#strange-pattern-on-a-flat-surface

closed 2019-08-06 by bubnikv