prusa3d/PrusaSlicer #2713: Print sequence

Pun-e, 2019-08-03:
All versions of PE-slic3r

Object order of standard Slic3r:
[image]

Order generated by Prusa slicer:
[image]

According to [this bug request](https://github.com/slic3r/Slic3r/issues/1184), this feature has been in slic3r since 2013, but I haven't been able to find out why it wasn't pulled into the prusa edition. This seems like a no-brainer and considering how much better Prusa's auto-arranging is I don't understand how this hasn't been taken care of for your own print farms.

cggorman, 2019-08-03:
Best I can tell, Pslicer prints in the order parts were added to the plate. That's fine, but I wish we could drag and re-order.

bubnikv, 2019-08-09:
> According to this bug request, this feature has been in slic3r since 2013

PrusaSlicer was forked from upstrem Slic3r after that date.

> Best I can tell, Pslicer prints in the order parts were added to the plate. That's fine, but I wish we could drag and re-order.

This only applies to the order of the instances. We shall improve on that.

And please, stop calling the application Plicer.

bubnikv, 2019-09-27:
I believe this is a duplicate of #375 #409 #1942, it will be fixed in PrusaSlicer 2.2. Closing.

closed 2019-09-27 by bubnikv