prusa3d/PrusaSlicer #2715: std: bad_alloc

maxminhtron, 2019-08-03:
### Version
PrusaSlicer 2.0.0+linux64-201905201652

### Operating system type + version

Linux 4.15.0-54-generic x86_64

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S 3D PRINTER

### Behavior
std: bad_alloc

I imported a 22 kbyte stl file and when I tried to slice it I get the error message :
std: bad_alloc.

Once I clicked ok on the error message box, the object mesh appears.

I tried to export the gcode, but would also get the same error message. 

The error message appears on the bottom  of the slicer window.

[image]

below is the stl file zip.

[Clock_by_A26_Remix_-_Minute_Hand.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3463934/Clock_by_A26_Remix_-_Minute_Hand.stl.zip)

_Is this a new feature request?_
 No

maxminhtron, 2019-08-03:
I renamed the .PrusaSlicer directory to something else, and start PrusaSlicer again and was able to slicer this object and larger ones without any problems.

bubnikv, 2019-08-05:
@maxminhtron Are you able to reproduce this bug yourself in any way?

maxminhtron, 2019-08-05:
Yes. I changed the settings and would get the same message. It may be the setting size.

bubnikv, 2019-08-05:
>  It may be the setting size.

You mean scaling of the object?

maxminhtron, 2019-08-05:
I am not sure. I tried different print settings, and would get that error message if "clock" is choosen.
I have exported the clock configuration and is included below:
[clock.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/3469345/clock.ini.zip)

lukasmatena, 2019-08-06:
@maxminhtron Thanks, providing the config helped. The cause of the crash is that you set infill extrusion width to 0% (Print Settings->Advanced), which is not the expected way how to turn off infill (I assume that's what you intended to do). Apparently, PrusaSlicer cannot cope and I get an ASAN crash during slicing (and the bad_alloc in case ASAN is disabled). PrusaSlicer should either not have allowed you to set it that way, or correctly understand the intent. We should do something about it, I'm not sure what. What do you think @bubnikv ?

Running in debug mode crashes even sooner (on loading the config), because of a failed assert in Flow::mm3_per_mm(). Variables at the time of the crash contain:
this->width = 0.0004
this->height = 0.1
this->bridge = false
res ends up being negative.

The log output and backtrace at the ASAN crash in Release mode:
> [2019-08-06 07:06:04.572013] [0x00007f200360e700] [debug]   Processing external surfaces for region 0 in parallel - start
[2019-08-06 07:06:04.575131] [0x00007f200220b700] [trace]   Processing external surface, detecting bridges. layer18.05, bridge groups: 2
[2019-08-06 07:06:04.579651] [0x00007f2002e0d700] [trace]   Processing external surface, detecting bridges. layer14.05, bridge groups: 1
[2019-08-06 07:06:04.584524] [0x00007f200360e700] [trace]   Processing external surface, detecting bridges. layer2.65, bridge groups: 1
[2019-08-06 07:06:04.584806] [0x00007f200220b700] [trace]   Processing external surface, detecting bridges - done
[2019-08-06 07:06:04.584843] [0x00007f200360e700] [trace]   Processing external surface, detecting bridges - done
[2019-08-06 07:06:04.586031] [0x00007f200260c700] [trace]   Processing external surface, detecting bridges. layer2.75, bridge groups: 1
[2019-08-06 07:06:04.590937] [0x00007f200260c700] [trace]   Processing external surface, detecting bridges - done
[2019-08-06 07:06:04.591047] [0x00007f2002e0d700] [trace]   Processing external surface, detecting bridges - done
[2019-08-06 07:06:04.591333] [0x00007f200360e700] [debug]   Processing external surfaces for region 0 in parallel - end
[2019-08-06 07:06:04.591423] [0x00007f200360e700] [info]    Discovering vertical shells... Resident memory: 1,290MB; Shared memory: 144MB; Private memory: 1,147MB; Peak memory usage: 1,292MB
[2019-08-06 07:06:04.591507] [0x00007f200360e700] [trace]   discover_horizontal_shells()
[2019-08-06 07:06:04.681511] [0x00007f200360e700] [info]    Bridge over infill... Resident memory: 1,301MB; Shared memory: 144MB; Private memory: 1,157MB; Peak memory usage: 1,301MB
[2019-08-06 07:06:04.705793] [0x00007f200360e700] [debug]   Filling layers in parallel - start
=================================================================
==2582==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7f04c53f2d58 at pc 0x7f050c40574d bp 0x7f04c53f2d40 sp 0x7f04c53f24e8
READ of size 8 at 0x7f04c53f2d58 thread T38
    #0 0x7f050c40574c  (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x5f74c)
    #1 0x1a5c592 in mbsrtowcs /usr/include/x86_64-linux-gnu/bits/wchar2.h:487
    #2 0x1a5c592 in wxMB2WC(wchar_t*, char const*, unsigned long) src/common/wxcrt.cpp:96
    #3 0x1a28d8a in wxMBConv::ToWChar(wchar_t*, unsigned long, char const*, unsigned long) const src/common/strconv.cpp:222
    #4 0x1a29b9f in wxMBConv::cMB2WC(char const*, unsigned long, unsigned long*) const src/common/strconv.cpp:403
    #5 0x1a32499 in wxString::ConvertStr(char const*, unsigned long, wxMBConv const&) src/common/string.cpp:388
    #6 0x1501ee9 in Slic3r::BackgroundSlicingProcess::thread_proc() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x1501ee9)
    #7 0x1502a4d in Slic3r::BackgroundSlicingProcess::thread_proc_safe() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x1502a4d)
    #8 0x7f050c0dcc7f  (/usr/lib/x86_64-linux-gnu/libstdc++.so.6+0xb8c7f)
    #9 0x7f05088d26b9 in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x76b9)
    #10 0x7f050860841c in clone (/lib/x86_64-linux-gnu/libc.so.6+0x10741c)
Address 0x7f04c53f2d58 is located in stack of thread T38 at offset 344 in frame
    #0 0x150097f in Slic3r::BackgroundSlicingProcess::process_fff() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x150097f)
  This frame has 6 object(s):
    [32, 64) 'export_path'
    [96, 128) '<unknown>'
    [160, 192) '<unknown>'
    [224, 256) '<unknown>'
    [288, 320) '<unknown>'
    [352, 576) '<unknown>' <== Memory access at offset 344 underflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism or swapcontext
      (longjmp and C++ exceptions *are* supported)
Thread T38 created by T0 here:
    #0 0x7f050c3dc253 in pthread_create (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x36253)
    #1 0x7f050c0dcdc2 in std::thread::_M_start_thread(std::shared_ptr<std::thread::_Impl_base>, void (*)()) (/usr/lib/x86_64-linux-gnu/libstdc++.so.6+0xb8dc2)
SUMMARY: AddressSanitizer: stack-buffer-overflow ??:0 ??
Shadow bytes around the buggy address:
  0x0fe118a76550: f2 f2 f2 f2 00 f4 f4 f4 f2 f2 f2 f2 00 f4 f4 f4
  0x0fe118a76560: f2 f2 f2 f2 00 00 00 00 f3 f3 f3 f3 f3 f3 f3 f3
  0x0fe118a76570: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe118a76580: f1 f1 f1 f1 00 00 00 00 f2 f2 f2 f2 00 00 00 00
  0x0fe118a76590: f2 f2 f2 f2 00 00 00 00 f2 f2 f2 f2 00 00 00 00
=>0x0fe118a765a0: f2 f2 f2 f2 00 00 00 00 f2 f2 f2[f2]00 00 00 00
  0x0fe118a765b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Heap right redzone:      fb
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack partial redzone:   f4
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
==2582==ABORTING

bubnikv, 2019-08-06:
The problem of this config is 
infill_extrusion_width = 0.4%
that is, there is an excessive percent, which leads to extremely small infill width numbers.

bubnikv, 2019-08-06:
This is quite a common problem, that one enters either a tiny number as a percent, or vice versa, enters an absolute value, but he things that the number was accepted as a percent. I know @YuSanka already implemented some safety checks in this regard. We should likely add more of these.

Also the application should not crash, that is for sure.

maxminhtron, 2019-08-06:
I did not changed the infil setting on purpose. It must have happen by accident. Thank you my friends. Now that the problem is resolved I will close the thread.

closed 2019-08-06 by maxminhtron

bubnikv, 2019-08-06:
The upcoming PrusaSlicer 2.1.0-alpha will produce some error message. Most likely you will get an error message just when you enter the value. If not, then you will get the following message when the configuration set is validated by the back end:

[image]

Zemistr, 2019-08-17:
Yeah... thanks to this, I'm not able to print 0.85 extrusion width with my 0.4 mm nozzle.
Is there a possibility to switch this off?

kazibole, 2019-08-18:
> Yeah... thanks to this, I'm not able to print 0.85 extrusion width with my 0.4 mm nozzle.
> Is there a possibility to switch this off?

Same here. Trying to print the collapsing lightsaber: https://www.thingiverse.com/thing:3606120

The comments indicate that you can print the vase mode blade files in vase mode, using an 0.85 mm extrusion width on an 0.4 mm nozzle. PrusaSlicer 2.1.0-alpha warns that this is an excessive extrusion width and will not slice. Going to try in PrusaSlicer 2.0.

cggorman, 2019-08-18:
Crazy thought...  Use a larger nozzle?

Zemistr, 2019-08-18:
Woooow... Why I wasn't thinking about it before??? 😑

I really don't want to change nozzle and calibrate the first layer every time when I need/want to print something with bigger extrusion width...
The printer is capable of doing it that way so why the Slicer is not allowing it?
I think at least in the expert mode it should be allowed.

bubnikv, 2019-08-19:
> Yeah... thanks to this, I'm not able to print 0.85 extrusion width with my 0.4 mm nozzle.
Is there a possibility to switch this off?

We need to set some reasonable threshold. I don't know how you can print such wide extrusions. Maybe your nozzle has a large flat area?

What limit do you recommend? What multiple of nozzle diameter?

bubnikv, 2019-08-19:
BTW leaving the parameters wide open potentially triggers some weird stuff inside slicer, producing weird bug reports to occupy us from writing useful stuff. That's why we added the check.

Zemistr, 2019-08-19:
I have a profile for "Vase mode" and there I have these parameters:
[image]

@bubnikv I understand. I'm using stock MK3 with 0.4 nozzle.

kazibole, 2019-08-19:
@Zemistr Have you measured your vase mode wall to see if it was actually 0.85 mm? I had done a quick test print using settings similar to yours, but my vase mode wall had only reached 0.78 mm.

Zemistr, 2019-08-19:
> @Zemistr Have you measured your vase mode wall to see if it was actually 0.85 mm? I had done a quick test print using settings similar to yours, but my vase mode wall had only reached 0.78 mm.

Good point... I will have a look and I will tell you 😉

FidelCapo, 2019-08-20:
#2784

francisco-pereira, 2019-08-20:
@kazibole  
I'm using the same settings and I'm getting 0.84~0.85mm. Did you compensate for filament width variations?

Zemistr, 2019-08-20:
> @Zemistr Have you measured your vase mode wall to see if it was actually 0.85 mm? I had done a quick test print using settings similar to yours, but my vase mode wall had only reached 0.78 mm.

I answered there ;)
https://github.com/prusa3d/PrusaSlicer/issues/2784#issuecomment-522898657

Schrunch, 2019-08-20:
I understand that there should be a limit but it could be lower. There is no problem printing an extrusion width equal to the layer height. I have tried things like printing 0.15 extrusion width with a layer height of 0.20 with a 0.20 nozzle and 1.2 extrusion multiplier and having the plate higher of 0.1 then the normal printing level (to have thiner width line then 0.20 before I found a 0.15 nozzle).
In Expert mode we should only have a warning in case we are asking something wrong by error.

bubnikv, 2019-08-21:
The limit was changed from 200% to 300% of the nozzle diameter.
Implemented with 7c0c570.

Pun-e, 2020-03-02:
> 
> 
> The limit was changed from 200% to 300% of the nozzle diameter.
> Implemented with [7c0c570](https://github.com/prusa3d/PrusaSlicer/commit/7c0c5705df1b6d7739c0d20b713e99741d91cc88).

I opened #3079 about this same issue, but it never received any attention.

The max should be 400% or a little over that (~410%) as that is the recommended extrusion width for maximal inter-layer bonding. This is based on information from one of my engineering textbooks, but I can't remember which one right now.

Would it be possible to bump it up from 300% to 400% so I can close my bug too?

Pro-tip: you can leave the outer perimeter at 100% to get parts that print dramatically faster, significantly stronger, and noticeably faster. I'm sure it will eventually become the accepted norm once it trickles down to the average user.