prusa3d/PrusaSlicer #2716: PrusaSlicer windows take forever to resize

jackw01, 2019-08-03:
### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows 10 1903

### 3D printer brand / version + firmware version (if known)
Custom built delta printer running Marlin

### Behavior
Open a PrusaSlicer window and resize it with the corner or edge resize handles. The window takes several seconds to respond to mouse input and only redraws about once every 10 seconds while moving the resize handle. My computer is fast and this does not happen with any software other than PrusaSlicer and Adobe Reader (it seems like other people have this problem with Adobe Reader). Moving the window does not cause this problem.

jackw01, 2019-08-06:
Nvidia 1050 Ti with GeForce Experience Driver version 431.60 (the latest version).

bubnikv, 2020-12-12:
Please try with prusaSlicer 2.3.0-beta3. Thanks.

jackw01, 2020-12-13:
Seems to be no longer an issue.

closed 2020-12-13 by jackw01