prusa3d/PrusaSlicer #2719: Using ONLY spool 5 (PETG) yields PLA temps.  Had to trick by falsely setting all spools to PETG.

MikeFaraday, 2019-08-04:
### Version
2.0.0+win64

### Operating system type + version
Win-10 Pro, version 1809, OS build 17763.615

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3 MMU2

### Behavior
I have been running PLA in MMU spools 1~4, and PETG in spool5.  When slicing a file using just spool 5, the G-code created contains temperature settings for PLA.  Setting the filament type to PETG, saving the Project, changing/loading any other Project, then reloading my actual (PETG) project shows PLA settings again used in the Filament tab.

I have a clumsy work around of setting all spools to Prusament PETG MMU2.  The fix is apparent via the Gcode default file naming as well as Gcode inspection.  It appears the software ignores what filament/spool is actually used in the design, and only relies upon what is set in spool1? (...guessing...))

The forum seems to have seen this issue long ago, and suggested it would have been fixed.
[[   https://github.com/prusa3d/PrusaSlicer/issues/648   ]]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2019-08-05:
I was unable to reproduce the issue. Could you please enclose a 3MF with a minimal example that I could open, slice and see that the gcode is wrong? I'm not sure I understand your description fully. Thanks.

antimix, 2019-08-10:
Did you remind to set Filament Type: **PET** for that spool, in **Filament Settings / Advanced / Filament Properties**  ?

lukasmatena, 2019-10-09:
This is similar to #2545.

I didn't manage to reproduce either of the issues and there is no response from their authors after a 3MF was requested. There also were two similar bugs, both fixed for 2.1.0 (commits 5393979 and 170bd8b), maybe this was flushed along. Could have also been a user mistake.

Closing.

closed 2019-10-09 by lukasmatena