prusa3d/PrusaSlicer #2720: How to call toolchange_gcode in the start gcode?

Schild0r, 2019-08-04:
Quick question:
I wanted to call the toolchange gcode (which has the parameter toolchange_gcode) in the end of my start gcode. But with 

> ...
> M190 S[first_layer_bed_temperature]; set bed temp and wait for temp to be reached
> M104 S[first_layer_temperature] T1; set nozzle temp for Extruder 2
> M109 S[first_layer_temperature] T0; set nozzle temp and wait for temp to be reached
> T0 G92 E0; reset extruder0
> G1 E20 G1 E0; prime and retract Extruder0
> T1 G92 E0; reset extruder1
> G1 E20 G1 E0; prime and retract Extruder1
> M117 Printing...; put message on display
> toolchange_gcode
> ;--------------------------------
> ;END OF STARTING SCRIPT
> ;--------------------------------

it will just put the plain string toolchange_gcode into the file
if i use 

> ...
> M117 Printing...; put message on display
> [toolchange_gcode]
> ;--------------------------------
> ;END OF STARTING SCRIPT
> ;--------------------------------

I will get a parsing error:

> Parsing error at line 16: Variable does not exist
> [toolchange_gcode]
>  ^
> !!!!! End of an error report for the custom G-code template start_gcode

so what am I doing wrong

Jebtrix, 2019-08-08:
You cannot call custom G-code templates from within another template like that in PrusaSlicer/Slic3r. You just need to copy **Tool change G-code** to your **Start G-Code**. 

The parameter name you see in the tooltip tells you what its called in the underlying .ini configuration file. The parameter names in the tooltip also helps if your going to be slicing via CLI.

Schild0r, 2019-08-08:
The problem is that you can't use [next_extruder] and [previous_extruder] in the start gcode

Jebtrix, 2019-08-08:
True. What exactly are you trying to do that you need [next_extruder] and [previous_extruder] variables in startup code? Logically only [next_extruder] in makes sense in startup context.

Schild0r, 2019-08-08:
> True. What exactly are you trying to do that you need [next_extruder] and [previous_extruder] variables in startup code? Logically only [next_extruder] makes sense in startup context.

Im having the problem that the toolchange retraction won't be applied on the first layer thus the nozzle which is unused at first will ooze onto the print until the first toolchange triggers the retraction (already did a post about that).

I figured out I might "code" my own toolchange retraction into the custom gcode where in the start gcode it would retract both extruders and then only unretract the [next_extruder] which is where the toolchange gcode comes into place 
(With the help of {if layer_num == 0 } I can skip the part that applies to the [previous_extruder]).
Luckily I noticed that the slicer will select the first used tool before applying the layer change gcode so I am doing the unretract thing in the layer change gcode with the help of {if layer_num == 0 } and the [current_extruder] placeholder which works in every gcode section.

Jebtrix, 2019-08-08:
I just read your other issue [https://github.com/prusa3d/PrusaSlicer/issues/2720](url). I think I get the gist. Why dont you just select the non current/initial extruder in Printer Settings > Start g-code

`T{if current_extruder == 0}1{else}0{endif}
or 
T{if initial_tool == 0}1{else}0{endif}
`

Schild0r, 2019-08-09:
Ist there a placeholder for [initial_tool]?
I assumed there are only the placeholders available that are mentioned in the slic3r manual. Are there more for PrusaSlicer?
This would of course solve many issues...

I don't think I could use [current_extruder] for this in the start gcode because I manually select the tools to retract thus the current extruder would be the last one retracted and not necessarily match the initial one

bubnikv, 2019-08-09:
initial_tool is the variable.
You may use the variable inside the curly braces {} inside any expression (condition, math etc)

Schild0r, 2019-08-09:
Cool cool
Is there a list for all available variables in PrusaSlicer?

bubnikv, 2019-08-09:
Please check the wiki of the github project.

closed 2019-08-09 by Schild0r

Schild0r, 2019-08-09:
Did not find the placeholder initial_tool there.
May you share a link where it is mentioned?

giPeteR, 2020-03-21:
I solved this with a Python post process code. Took a week or two, but now it's great.
https://github.com/prusa3d/PrusaSlicer/issues/2710#issuecomment-602056147