prusa3d/PrusaSlicer #2721: Can't change which MMU2 channel selected for model on bed

rogerfroud, 2019-08-04:
### Version 2.0.0+win64 Windows 10 V1903
MK3S + MMU2S

I've got my MMU2S set up with PLA in channel 1, PETG in channel 2 and PVA in channel 5
I'm printing a single colour item in PETG with soluble supports, but there doesn't seem to be a way to tell the slicer that I want to use channel 2 instead of channel 1. I've watched a video of an old version of the slicer, and it looks like you used to be able to tell it what channel to use for each of a multi-colour print. I don't know if you can still do that with more than one colour as well as the soluble one.

I would have thought you could just right click on the model and select which channel you want from the list. Maybe it is possible, but if it is, it's far from being intuitive. I've looked everywhere and I can't see how to do it.

lukasmatena, 2019-08-05:
> I would have thought you could just right click on the model and select which channel you want from the list.

This is exactly the way to go, if you click the object in the scene. Another is to select the extruder in the object list, when you left-click the extruder number (for that you have to be in Advanced or Expert mode).
[image]

If you want to use some specific extruder for support / support interface, go to Print Settings->Multiple Extruders and set it the way you want. You most likely want to set support material extruder to 0 and support material interface extruder to 5 (it is only possible if you set support material contact distance to zero (Print Settings->Support material->Contact Z distance)).

rogerfroud, 2019-08-05:
Ok, thanks for that. I guess that was off the screen when i was looking for
it. Perhaps it might be a good idea to put it further up since you can get
to all of the other options through other means, but this is the only way
to change the extruder.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=oa-4885-b>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=oa-4885-b>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bubnikv, 2019-08-09:
Issue resolved, closing.

closed 2019-08-09 by bubnikv