prusa3d/PrusaSlicer #2723: need placeholder for firmware version (used in start gcode M115)

draeath, 2019-08-05:
### Version
2.0.0

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
Stock Prusa I3 MK3S from kit

### Behavior
If you modify your extruder profile to customize start gcode, the 'M115' command bakes in (to the profile) the firmware version that is current at this time. I am requesting that a new placeholder (such as `[first_layer_bed_temperature]`) be added to Prusa Slicer, so that can be used instead of a string value for the M115 command. This allows continual use of custom start gcode through time without the user having to take care to keep updating the M115 value. This in turn reduces the chances of missed firmware updates.

This is a new feature request, albeit one under the hood and not any new UI functionality.

I would create a PR, as the placeholder itself would not be difficult to add. Howevr, the complexity comes in determining where the Slicer will retrieve it's value from at slicing time.

#### Project File (.3MF) where problem occurs
N/A

bubnikv, 2019-08-06:
We have already discussed this feature internally. It certainly makes
sense, it will simplify maintenance of our profiles. It will likely not be
part of PrusaSlicer 2.1.0 yet though.

draeath, 2019-08-06:
You might be able to leverage the github public API and have the slicer check for the latest release tag, though this is "unfriendly" in that it throws clients at github. You'd certainly need to mind how frequently it checks, caching etc. That's the real sticky bit, actually consuming it in the slicer is likely to be easy compared to it.

Additionally if firmware versions for printers (eg MK2 vs MK3 vs any+MMU) begin to drift, it complicates that further. The "proper" way would involve building a basic API on the Prusa webstack that provides firmware versions for different printers, and the slicer figures out which one is appropriate based on the selected profile.

xunleii, 2020-12-13:
I would like to bump this old issue with a new idea; instead of using an API on the Prusa webstack to know if a new Prusa firmware is available, why not just using the firmware version value available in default start gcode ? 

This value is "automatically" updated when Prusa Slicer is updated, so we can use a placeholder containing this value.

mrjackson, 2020-12-13:
> This value is "automatically" updated when Prusa Slicer is updated, so we can use a placeholder containing this value.
This only true for a default printer profile, if you make your own (to save the config for Octoprint for instance), it does not update anymore.

xunleii, 2020-12-14:
> > This value is "automatically" updated when Prusa Slicer is updated, so we can use a placeholder containing this value.
>
>  This only true for a default printer profile, if you make your own (to save the config for Octoprint for instance), it does not update anymore.

Yep, this is the reason why I'd like to have a placeholder for the firmware version; for each new firmware released, we can just update the placeholder value instead of updating all default profiles. In addition, this placeholder can be used by us on our custom printer settings (in that way, the version checked will be always up-to-date).

closed 2026-09-29 by prusaslicer-bot[bot]