prusa3d/PrusaSlicer #2724: PrusaSlicer Website ends up with 404

DominikPalo, 2019-08-05:
### Version
2.0.0+

### Operating system type + version
macOS Mojave 10.14.5

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
1) Open the "Help" menu
2) Click on the "PrusaSlicer Website" item
* _Expected Results_
The PrusaSlicer website is opened
* _Actual Results_
A browser is navigated to "https://www.prusa3d.com/slicerweb&lng=sk_SK" which fails with a 404 error
[image]

bubnikv, 2019-08-06:
We are working on it. The Prusa3D hosted web pages expect explicit set of locale identifiers, for which we localize the web pages. We do not translate to Slovak yet, so I have just added a mapping from Slovak to Czech with 621a552dc0bce535d7b11d47770f593167be39e3. I hope that will do for now.
The change will be part of the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-06 by bubnikv

DominikPalo, 2019-09-27:
I've just tried the latest version (2.1.0) and it opens the [en_US web](https://prusa3d.com/slicerweb&lng=en_US), while the OS language is still set to Slovak. So it is definitely better than 404, but it seems that your mapping to the Czech version doesn't work. 🤔

FidelCapo, 2019-09-30:
Can I ask you in what language do you have a slicer?

DominikPalo, 2019-09-30:
My slicer is in English

FidelCapo, 2019-09-30:
That's why the websites opens in English.

DominikPalo, 2019-09-30:
@FidelCapo I was always using Slicer in English, so:
1) Why the older versions (e.g. 2.0.0) were opening the sk_SK version of the page (I thought that is because my OS is set to SK language)
2) Slicer doesn't contain SK language... so why the commit 621a552 introduced the `sk_SK` to `cs_CZ` mapping? When is that mapping used?

FidelCapo, 2019-09-30:
This mapping is used when you start Prusa Slicer for first time. It's new from 2.1.0 so older versions can work differently.