prusa3d/PrusaSlicer #2725: Losing parts of the stl when slicing and printing

mowermanjp, 2019-08-06:
### Version
2.0.0

2.0.0+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10 Pro
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_Prusa MK3S / Stock, 3.7.2-2363

### Behavior
* _Details missing after slicing, stl file verified working in Cura 4.2
* _Slice and print
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]
[image]
Notice the center of the 4
_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

YuSanka, 2019-08-07:
Thanks for your report. Could you please send for us your model.

mowermanjp, 2019-08-07:
MK3S

neophyl, 2019-08-07:
no they need a zipped up copy of your .3mf project file.  That way they can try and duplicate the issue.

rtyr, 2021-02-03:
No respone. Closing.

closed 2021-02-03 by rtyr