prusa3d/PrusaSlicer #2726: Prompt to save part / modifier changes

GWdd, 2019-08-06:
### Version
2.0.0

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
Pr Mk3/S

### Behavior
Plicer will prompt to show changes have been made to simpler settings in profiles, settings, etc, and generally these are minor and easily repeatable changes (infill 15% to 10%).  And often aren't needed to update the original profile, and can be as banal as selecting PLA instead of Prusa PLA.

But loading a project, or starting a new project, .3MF,  then making changes to the bed; adding, sizing and positioning modifiers, enforcers, and adjusting part positions can take a large effort that a person really doesn't want to replicate.  But all this work gets dumped into the trash bin if they close the app before saving the project, and there is no warning the project has changed, users only get the same notification "something in a profile has changed".  

How about a warning that the BED has been modified when exiting the app.  Something to let users know significant changes have been made.  

_Is this a new feature request?_  Probably - and may even be a duplicate, I haven't done an exhaustive search for similar requests.

#### Project File (.3MF) where problem occurs
Pick one ...

bubnikv, 2019-08-09:
May I ask you guys to stop call this application Plicer? At least to me, it sounds offending. Please try to invent another shortcut.

GWdd, 2019-08-09:
Prusa Slic3r is too long to type every time the name is used.  It's like typing out fernserhapparat instead of TV,  PS is already used (Photoshop).

Suggest a nickname you like.

bubnikv, 2021-06-03:
The longest German word ever is known to be [Donaudampfschiffahrtselektrizitätenhauptbetriebswerkbauunterbeamtengesellschaft](
https://en.wikipedia.org/wiki/Donaudampfschiffahrtselektrizit%C3%A4tenhauptbetriebswerkbauunterbeamtengesellschaft#:~:text=According%20to%20the%201995%20Guinness%20Book%20of%20World,%28%22electricity%22%29%20is%20normally%20used%20only%20in%20the%20singular.)

Concept of a project state has been implemented by @enricoturri1966 for PrusaSlicer 2.4.0-alpha1. 
Closing.

bubnikv, 2021-06-03:
https://en.wikipedia.org/wiki/File:De-Donaudampfschiffahrtselektrizit%C3%A4tenhauptbetriebswerkbauunterbeamtengesellschaft.ogg

closed 2021-06-03 by bubnikv