prusa3d/PrusaSlicer #2728: PrusaSlicer 2.0 - Plater Icons Not Visible on Dual Monitors

robo67, 2019-08-07:
### Version
PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 15063), 64-bit edition

OpenGL installation
GL version:   1.1.0
Vendor:       Microsoft Corporation
Renderer:     GDI Generic
GLSL version: N/A
Installed extensions:
GL_EXT_bgra
GL_EXT_paletted_texture
GL_WIN_swap_hint

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 - fw 3.7.2

### Behavior
I have a dual monitor configuration using a DisplayLink docking station.  When I launch PrusaSlicer when my laptop is attached to the docking station none of the icons display in the Plater tab - see attached screenshot (PrusaSlicer-GUI-Issues).  I am only using the external displays - my laptop is closed and docked.
[image]

When I launch PrusaSlicer using only my laptop display, the icons display without issue - see attached screenshot (PrusaSlicer-GUI-OKAY)
If I then reconnect to my dual display, the icons are still okay.
[image]

The problem happens only when connected to my dual display.  I have updated drivers for the DisplayLink and the issue keeps occurring. 

#### Project File (.3MF) where problem occurs
None.  The problem is with the application.

bubnikv, 2019-08-08:
You answered yourself above:

OpenGL installation
GL version: 1.1.0
Vendor: Microsoft Corporation
Renderer: GDI Generic

For some reason you are not getting a hardware accelerated OpenGL context
when running on the other monitor.

robo67, 2019-08-08:
It must be an issue during boot.  If I boot PrusaSlicer whilst connected to my dual monitors I get the problem.  If I boot using only my laptop display and then connect to the dual monitors, all works well.

What could be causing the problem?

Thanks,
Alan

PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 15063), 64-bit edition

OpenGL installation
GL version:   4.3.0 - Build 20.19.15.5063
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 4600
GLSL version: 4.30 - Build 20.19.15.5063
Installed extensions:

GL_3DFX_texture_compression_FXT1
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
[... 128 lines cut ...]
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_IBM_texture_mirrored_repeat
GL_INTEL_fragment_shader_ordering
GL_INTEL_map_texture
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_debug
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint

Confidentiality Notice: This message (including attachments) is a private communication solely for use of the intended recipient(s). If you are not the intended recipient(s) or believe you received this message in error, notify the sender immediately and then delete this message. Any other use, retention, dissemination or copying is prohibited and may be a violation of law, including the Electronic Communication Privacy Act of 1986.   ­­

bubnikv, 2019-08-09:
I don't know. Windows? The device driver provided by your graphics card
vendor?

bubnikv, 2019-08-09:
We shall add a test for the OpenGL context. Slicer is trying to load the software renderer library Mesa in case the system driver does not offer OpenGL 2.0 and newer context, but some system OpenGL drivers cannot be unloaded. We should at least add some diagnostics of this issue.

bubnikv, 2019-08-23:
PrusaSlicer supports OpenGL 2.0 and newer. The upcoming PrusaSlicer 2.1.0-beta will provide an appropriate error message with an advice how to execute PrusaSlicer with software rendering.
Closing.

closed 2019-08-23 by bubnikv

Strooom, 2020-04-19:
I am having the exact same problem.
As USB port replicators with external monitors are becoming more popular, I think it is worthwhile to investigate this, especially as PrusaSlicer seems to work ok after starting it undocked and then docking to activate the external displays

justinbarcelos, 2020-04-26:
Yes, this issue needs to be reopened. I have the exact same problem when using my USB-C dock that runs my external monitors.
[image]

ITGuruWoo, 2020-10-01:
I am having the same problem. Was this ever fixed? Thanks