prusa3d/PrusaSlicer #2729: Feature Request:  FAST Skinnydip string removal.

TNDavid, 2019-08-08:
I have been using Erik Bjorgan's Skinnydip python Script for some time now, as it is probably one of the best approaches to removing strings when using MMU2. However, it adds a **_LOT_** of time to the print if the temperature reduction feature is used.

I have been doing a lot of testing over the past couple weeks to evaluate my theory that reducing the temperature only for the dip cycle works just as well as reducing it for the filament retraction, without adding any or at least minimal time to the print. Skinnydip was developed to remove the strings, so reducing the temp before retracting is of little use, since the dip cycle is to remove the string anyway.

My setup is a MK3 with MMU2. Testing done with PETG, 240 deg print temp, 225 deg dip temp.  Other settings are 28mm cooling tube position, 15mm cooling tube length.  It is working very well. 

The two main problems for MMU2 users are filament strings, and increasing tip diameters over time as the PTFE tube enlarges. I developed a Torlon Tube insert to get consistent tip diameters over time, https://www.facebook.com/groups/prusacommunity/permalink/931436870530379/

I am currently using the Skinnydip python script for string reduction, but as Skinnydip stands currently if one uses the temperature reduction considerable time is added to the print cycle, far too much for a print with a high number of filament changes. Also, the script does not initiate the dip cycle for the last filament used, so that one will have a string and cause problems on the next print.  In the attached modified G-code file I added the dip cycle to the end for the last filament used.

I first tried dipping a shorter distance than the initial retraction, but that did not have the effect I originally imagined.

I then added the temperature reduction for the dip cycle only by modifying a G-code file for a single layer, five filament print. The modified file lowers the temperature during the cooling cycle, with a wait at end of cooling to insure temp has reached the lowered setting, so depending on the cooling time and time to reduce the temperature this may or may not add any time to the filament change. I am using a silicone sock on my hot end, so temp reduction takes longer than one without the sock, but with my 15mm cooling tube length at 1mm/sec it is very close to the same time, so little to no time added. The temp return to printing temp takes place during the filament change, so no added time.

The greatest impact on Skinnydip string removal turned out to be the pause time at the bottom of the dip! I first had no pause, but that left short wrinkled strings, no time to melt them off. I then tried 200 msec, but that reproduced 4-8mm long strings, time was long enough to remelt the tip end. 100 msec improved it, 50 msec seemed to be the best.

Like Erik, I am hoping that Skinnydip can be added to Prusa Slicer as an OPTION for users, with settings for dip temperature, dip length, and dip pause time. Skinnydip WORKS, but the time addition for temp change as it currently exists is far too great for a print with a high number of filament changes. This modified method works just as well, with little to no time addition. 

I am attaching my single layer five filament test G-code text file, hoping that others will look it over and comment, or even better test it to make further modifications or suggestions to further improve Skinnydip to make it viable to add to Prusa Slicer for a fast and simple option for string removal. This is far simpler than trying to modify ramming, which is very complex, with little documentation to show the results of different settings.  Skinnydip is EASY for users, and with this modification to it adds little to no time to the print cycle.
(https://github.com/prusa3d/PrusaSlicer/files/3481906/Fast.Skinnydip.Single.Layer-5.Filament.txt)

antimix, 2019-08-10:
@TNDavid , I used a **Skinnydip** implementation in PrusaSlicer, but I was not satisfied by the results.
So I changed methodology, and I created from scratch **a new system that I called "Dribbling"** that use the filament as a bouncing ball until the filament has almost no string.  
Currently I do not change the temperature (I would have lowered it, since the results are better)  because it increases the time, and because now there is a good compromise between the speed and quality. After a lot of test and several change of mind, I choose a code logic that resulted the best. 

I have already implemented **_"Dribbling"_** it in [**my fork of PrusaSLicer 2.1.0-alpha1,**](https://github.com/antimix/PrusaSlicer/tree/Dribbling_v.2.1) and today I have done my first MMU2 multicolor print without a single intervention. It never ever happen to me, that I was able to obtain a smooth MMU2 multicolor print without any failure. 
However I have also changed my Prusa I3MK3S replacing the PRUSA heat-break (with double diameters) with the original E3D heat-break (with a thin diameter) and this avoid the formation of "big" filament heads.

If you are interested I can prepare a big ZIP with the executable.

TNDavid, 2019-08-10:
I would like to test that too, however my test with the FAST implementation of a Skinnydip file worked great, with little to no added time.  Unfortunately I cannot do the coding necessary beyond modifying a 
g-code file.

antimix, 2019-08-11:
@TNDavid did you try the Skinnydip integrated in PrusaSlicer ? 
You can find [HERE](https://github.com/domesticatedviking/PrusaSlicer)
May be it works for you, and you can see the difference between the post processor results and the direct results. Let me know. It did not worked for me, but as you know every printer and filament set is different and may require different approaches.

TNDavid, 2019-08-11:
Yes.  It works the same as the script. The problem with it is that if temperature reduction is used it adds a LOT of time, like the skinnydip script, since it waits for the temperature to drop before retracting, adding that wait time to every filament change.  I did considerable testing of the original script before Erik developed the integrated one.  My change in the g-code file was to cool down the hot end during the retraction and cooling moves, for little to no time cost.  Skinnydip WORKS, but unless this "fast" modification is made to it, the temp reduction feature is far too time consuming.
I use the Skinnydip script for multi material printing, as it is one of the only things many of us have found that makes the MMU2 reliable.  Strings are a real issue, and with my Torlon tube insert this is even worse, since the Torlon does not stretch out and get larger, the reason for the Torlon development.  If the string folds over, it either stretches out the PTFE, making it larger over time, or needs higher force to push through the Torlon. I run Skinnydip, and can print very reliably with the MMU2, but the temp reduction helps a little more, but will not use it because of the inordinate time penalty.  
The only reason for my mod is time reduction.  It saves about 15-20 seconds per filament change, which is huge for a print with many hundreds of filament changes.
[image]

antimix, 2019-08-11:
@TNDavid I see. I never tried mine with PET. I designed it with the goal of PLA.
Those are the tip I get. I should try with PETG and see what happens. May be I will be forced to insert also an optional Temperature values.
[image]

WarrenSchultz, 2019-08-15:
@antimix 
> @TNDavid , I used a **Skinnydip** implementation in PrusaSlicer, but I was not satisfied by the results.
> So I changed methodology, and I created from scratch **a new system that I called [...]

I'd be interested to try an executable build. I've tried the skinnydip variant of prusaslicer, and it was too buggy for me to use effectively. Thanks!

tsamisa, 2019-08-29:
@antimix
I used the prebuild version of skinnydip and in my case it gave me the first sucessful print on my MMU2S prusa. I had to reduce the pause parameter to reduce the thickness of the filament heads. I used PTEG. The problem is that the prebuild version, for some reason, crashes when i try to load 3mf/amf files and building from source with VStudio 2019 was a PITA. I would like to give your "Dribling" version a try also if you have a pre-build version for windows. I dont think i'll change the heatbrake. Is this really a must? 
Finally both of these approaches seem to drastically improve MMU2S loading problems. They work in a similar concept and i'm really wondering why prusa doesn't seem to be willing to incorporate it in their slicer. The codes and logic are already there, users feedback is mostly positive, and can be refined by prusa and added as a optional setting. I know there is a lot of extra time in loading/unloading but **TNDavid's** logic seems solid in reducing time and having to print a 12 hour model just to find out that when you just left the house there was a failure it takes a lot more time on the long  run.

antimix, 2019-08-30:
@tsamisa At the moment I do not care if Prusa implement those changes,
since they are focused on bigger and more important functionalities.
In the coming days I will synch my version with the latest Prusa version to
incorporate all my changes to it, and the result is a Prusa version with
dribbling changes.

rmilyard, 2019-09-15:
Guys sorry but I am new to the MMU2S.  I have a MK3S with new Bontech Mosquito hotend.  I have been having a lot of problems with jamming in multi-color.  Single color prints fine.  I am really thinking it's the tips.  I would love to try something that might help.  How can I get executable version this PrusaSlicer with these changes?

tsamisa, 2019-09-16:
@rmilyard
my setup is mk3s mmu2s with bmg bondtech. There is a prebuild version of prusaslicer with skinnydip. There is a post with a link in this discussion https://github.com/prusa3d/PrusaSlicer/issues/2385. I havent used the dribbling method yet but skinnydip made my mmu2s workable. The problem is the added delay and that the prebuild version has a problem loading 3mf/amf files. But i think the refined approach of TNDavid and the dribling of antimix are another step forward towards better tips.

rmilyard, 2019-09-16:
> @rmilyard
> my setup is mk3s mmu2s with bmg bondtech. There is a prebuild version of prusaslicer with skinnydip. There is a post with a link in this discussion #2385. I havent used the dribbling [...]

I am wanting to just be able to print something mutli color with this combo.  I almost always get jammed at first color change.  Looking at the post but didn't see built version yet but just got home from work so kinda out of it right now.  LONG DAY!

tsamisa, 2019-09-17:
@rmilyard 
The link is still in that post. 
https://mega.nz/#!b3I3kYBT!ybRMhsaUdL1NJUiPGQE2U3SMaEWA6oQwC1-xLjuixLQ
I know what you mean with mmu2s. At best case scenario i could go up to 5-6 filament changes before getting loading failures. I removed the buffer, printed various components found in forums and thingverse. Put my filament in boxes, out of boxes, over mmu2 next to it. But nothing did the job. I'm using mostly petg. After some tweaking with skinnydip (basically the pause option) i got near perfect tips and for the first time i managed to print a successful multicolour print.

rmilyard, 2019-09-17:
@tsamisa 
Thanks for the link.  I will download it and check it out.  So your setup is like mine but are you also using Slice Mosquito?  I print PLA and PETG.  Mostly PLA for multi color since I have a bunch.  If wouldn't mind could you share your settings for a starting point for me to try?

m3cinc, 2019-09-17:
All I can state at this point is that for PETG multi filament print, Skinnydip is an absolute must. I was never able to print reliably until I used the Skinnydip slicer version. Can just hope this feature will make it in the next release.

tsamisa, 2019-09-18:
@rmilyard 
I don't have the mosquito. I have the bmg bondtech extruder upgrade for the prusa. I use it because it resolved a lot of issues when printing with 0.25 nozzle and 0.5-0.7 layer height. With prusa extruder, even with a heatsink and fan attached to it, at some point there was a heat creep  causing layer skipping. I dont know about the mosquito version, but with mine i had to change the filament parking position to 80 and Extra loading distance to -6.5 (Found under Printer Settings->Single extruder MM setup).
Also for the skinnydip (Prusament PETG) for me what works is: 
Toolchange temp:230
Insertion distance:32
Pause: 15
Speed to move to melting zone: 33
Speed to retrect: 70

tsamisa, 2019-09-18:
@m3cinc 
I absolutely agree but it seems that it didnt found its way to the latest prusaslicer. But as i said personally i would really like to use **TNDavid's** refinements and try **antimix's** dribbling method also.

antimix, 2019-09-19:
As I wrote in the forum, probably after the PrusaSlicer 2.1 final version release, I will upload the windows executable of "Dribbling PrusaSlicer 2.1"
Upon great interest, and a lot of successfully results on usage by several users may be I will also upgrade Skinnydip to the latest PrusaSlicer 2.1 and will put it on my repository.
I would have preferred that the original Skinnydip owner would have updated it on its legitimate repository, but the opensource world is nice also because everyone can contributes.

rmilyard, 2019-09-20:
> As I wrote in the forum, probably after the PrusaSlicer 2.1 final version release, I will upload the windows executable of "Dribbling PrusaSlicer 2.1"
> Upon great interest, and a lot of successfully [...]

I would love to try it out. Just want something that works more than not. At this point my Mosquito MMU2S combo isn’t working out.

tsamisa, 2019-09-20:
@antimix 
Thanks for all the effort you put into this

antimix, 2019-09-21:
@TNDavid 
I have ported Skinnydip to the latest PrusaSlicer 2.1.0 😄
Now, just to understand better what you were explaining, since I am not sue to have well understood your issue, I have some questions.

When you say that cooling down with wait has big impact on the print time...
> The problem with it is that if temperature reduction is used it adds a LOT of time, like the skinnydip script, since it waits for the temperature to drop before retracting...
I agree with you.  On my first Dribbling I had temperature changes, and during the tests I notice that the impact on time was unacceptable for a lot of changes, so I decided to not use temperature changes in my Dribbling.

- Regarding Skinnydip you seem to propose to change the temp without the WAIT parameter, correct ?
So from your file:
`;************************************************`
`M400 ; Wait for cooling moves complete to change temp`
`M109 R225 ;***SKINNYDIP Waiting for maximum target temperature = 225, waits until cooling`
`M104 S240 ; Return to print temp -> Set the temperature of the current extruder to 240C and return control to the host immediately (i.e. before that temperature has been reached by the extruder)`
`;*****SKINNYDIP THREAD REDUCTION*****************`
`G1 E28 F2000  ;move stringy tip into melt zone`
`G4 P50        ;pause in melt zone`
`G1 E-28 F4000  ;extract clean tip from melt zone`
`; ***********************************************`

- I see that as soon as the hotend is cooled, you immediately set the new high temp and before it can really become hot it immediately does the skinny moves. Correct ?
This is an actual Skinnydip 2.1.0 excerpt

**M109 S200** ;SKINNYDIP TOOLCHANGE WAIT_FOR_TEMP
G1 E-15.0000 F6000
G1 E-24.5000 F4800
G1 E-7.0000 F2400
G1 E-3.5000 F1440
M104 S198  ;RESTORE PRE-TOOLCHANGE TEMP
M104 S190
G1  Y135.480
G1  X173.216  E20.0000 F3000
G1  X218.670  E-20.0000
G1  X177.004 E20.0000
G1  X215.465 E-20.0000
G1  X179.751 E20.0000
G1  X213.084 E-20.0000
SKINNYDIP START
G1 E31.0000 F1980
G4 P50
G1 E-31.0000 F4200
G4 P0
; SKINNYDIP END

I see that only the first command needs to be changed. Correct ?

antimix, 2019-09-21:
You will find  the version of **SKINNYDIP** and **DRIBBLING** for PrusaSlicer 2.1.0 at
[https://github.com/antimix/PrusaSlicer/releases](https://github.com/antimix/PrusaSlicer/releases)
Le me know.

rmilyard, 2019-09-22:
> You will find  the version of SKINNYDIP and DRIBBLING for PrusaSlicer 2.1.0 at
> https://github.com/antimix/PrusaSlicer/releases
> Le me know.

So what is a quick run down of the differences?  I am not sure which would work best for my combo of MK3S+MMU2S with Bondtech Mosquito.

tsamisa, 2019-09-22:
@antimix 
Well the first test with the dribbling method showed very positive results from Prusament PETG. I have a small test file with only 12 changes and with 3 drib moves completed with no problem. I Did you apply TNDavids  change in the SKINNYDIP version? I'm just very curious to compare times. But for me both method are a must.
[image]

rmilyard, 2019-09-22:
So am not sure which program I should try and what settings to try for my setup.  Sorry but new to MMU2 and these mods.

tsamisa, 2019-09-22:
@rmilyard
Try both and decide what suits you. I already replied to you  with the settings i use for skinnydip. Its for PETG.

antimix, 2019-09-22:
@tsamisa 
No, Skinnydip is exactly as it was. I did not changed it becouse I did not understand exactly what it needed.

@TNDavid 
Can you give me a block of GCODE output generated with my latest Skinnydip 2.1.0, and then the same piece, manually "corrected" with the changes you expect to be done automatically by Skinnydip ?

TNDavid, 2019-09-22:
> @TNDavid

> * Regarding Skinnydip you seem to propose to change the temp without the WAIT parameter, correct ?

**_Yes, Start the temp change, but proceed to the cooling moves. The temp drop take place during the retraction and cooling moves, so that time is not wasted.  My thought is that the lower temp is only helpful for the dip, since the dipping purpose is to remove the string anyway.  I moved the wait to the start of the dip cycle to insure that the lower temp has been reached before the dip.  My settings for PETG are 28mm cooling tube position, 15mm cooling tube length, 240 printing temp, 215 temp for dip cycle, 50 msec pause in melt zone.  (I tried multiple pauses, over 100 msec just made a new string, no pause folded the string up but did not melt it all off). the dip insertion should be the same as the cooling tube position, or slightly less. If set longer the cooled tip is rammed back into the melt pool._**  

****>   So from your file:
> **`;********************************************`**
>   `M400 ; Wait for cooling moves complete to change temp`
>   `M109 R225 ;***SKINNYDIP Waiting for maximum target temperature = 225, waits until [...]

_**Yes, the dip cycle is so fast that the temp change being prior has no time to actually raise the temp before the dip cycle has finished.**_

>   This is an actual Skinnydip 2.1.0 excerpt
> 
> **M109 S200** ;SKINNYDIP TOOLCHANGE WAIT_FOR_TEMP
***************************
**_I do not understand all these extruder moves, seem to be extrnious_** 
> G1 E-15.0000 F6000 
> G1 E-24.5000 F4800                            
> G1 E-7.0000 F2400                        
> G1 E-3.5000 F1440
**************************
> M104 S198 ;RESTORE PRE-TOOLCHANGE TEMP
> M104 S190
> G1 Y135.480
> G1 X173.216 E20.0000 F3000
> G1 X218.670 E-20.0000
> G1 X177.004 E20.0000
> G1 X215.465 E-20.0000
> G1 X179.751 E20.0000
> G1 X213.084 E-20.0000 [...]

_**I did not just remove the wait, I moved it as a M400 to just before the dip. In my PETG settings, with two cooling moves, 1mm/sec first move 2mm/sec second move, the time works out that there is little to no wait since the temp has reached the lowered set point by the time the retraction and cooling have finished. I use a silicone sock, so if one not used temp reduction will be even faster.**_

rmilyard, 2019-09-23:
So guys is anyone using this combo and have either Dribbling or Skinnydips working right?  MK3S+MMU2S with Bondtech Mosquito?  I am having issue each time filament changes,  It seems to be strings on tips or tips to big.  Single color prints fine.

Sorry to ask but I am getting lost trying to figure this out.  I am using custom firmware from Vertigo for this extruder and slice thermistor.

m3cinc, 2019-09-24:
With skinnydip, I am now running ~600 filament changes without issues. The
key is to print PETG at minimum+5C and to drop the temperature by at least
10-15C before removal, for example, with Prusament PETG, I am setting up
like this:
[image: image.png]
[image: image.png]
Not all PETG have saame glycol content and thus the temperature can be
different. For example PETG from Stronghero works best with:
[image: image.png]
[image: image.png]

Hope this helps...
Marc

rmilyard, 2019-09-24:
> With skinnydip, I am now running ~600 filament changes without issues. The
> key is to print PETG at minimum+5C and to drop the temperature by at least
> 10-15C before removal, for example, with [...]

Missing images.

TNDavid, 2019-09-24:
> So guys is anyone using this combo and have either Dribbling or Skinnydips working right? MK3S+MMU2S with Bondtech Mosquito? I am having issue each time filament changes, It seems to be strings on [...]

Yes, I have been using theSkinnydip Python script for some time now, the only way I have been able to use the MMU2 for PETG and get good results. I am using the stock E3D V6 hotend.  However, I do not use the temp reduction, since it waits for the temp to reach set point before retracting and doing the cooling moves, adds a LOT of time to a print with many filament changes.  Awaiting a Prusa Slicer  mod to add my Fast method of Skinnydip.  That will give the best of both worlds, Skinnydip, Temp reduction, and little to no time lost.

rmilyard, 2019-09-24:
So I tried using PrusaSlicer 2.1.0 with Skinnydip integrated with default settings.  I am using PLA.  It worked better.  I can change filaments a few times then load fails.  It has little hair and end little bigger.  When I cut it off and keep going loads fine.  So guessing still need little fine tuning but not sure what should be changing?

m3cinc, 2019-09-24:
Typically you want to set the cooling temp ~10C below the extrusion
temperature. That works for PETG and also for PLA.

Siwat2545, 2019-09-25:
I use the bmg with the mosquito Magnum, if I don't use the script the mmu load fail happen but if I use it I get jam in the heatbreak , this happen with pla, but with petg I have some jam but not as much as pla

tsamisa, 2019-09-25:
@Siwat2545 I'm using mostly PLA. I found that what made a difference in forming the tip's thickness, in the skinnydip version, is the Pause setting. In my case a _Pause_ of 15 ms was enough. I think is a good idea to play with that setting. Although I'm using the bondtech extruder with the E3D v6 Prussa heatbrake and not a mosquito setting.

rmilyard, 2019-09-25:
> @Siwat2545 I'm using mostly PETG. I found that what made a difference in forming the tip's thickness, in the skinnydip version (This whole phrase sound kind of kinky :) ), is the Pause setting. In my case a Pause of 15 ms was enough. I think is a good idea to play with that setting. Although I'm using the bondtech extruder with the E3D v6 Prussa heatbrake and not a mosquito setting.

Which pause?  melt zone or before extraction?

rmilyard, 2019-09-25:
So better.  Had failed load 3 times but much better.  Tips still look stringy to me.  Please look at pictures in link and let me know what you think.

https://share.icloud.com/photos/0mnMYe0LjmO_m81JAcFBDfUrQ

tsamisa, 2019-09-25:
@rmilyard 
Try your pause in the melting zone. You have to find the time needed to melt those strings. I would advice you also to try the dribbling method also. In my case with 3 dribbling moves it seemed that it was a good approach also.

rmilyard, 2019-09-25:
> @rmilyard
> Try your pause in the melting zone. You have to find the time needed to melt those strings. I would advice you also to try the dribbling method also. In my case with 3 dribbling moves it seemed that it was a good approach also.

I have both Programs downloaded.  I have been messing with Skinnydips.  So with Dribbling version what all sets mess with there?  Just turn on dribbling and set for 3?

antimix, 2019-09-25:
Yes, because Dribbling and Skinnydip use separate and different parameters.
Each one can handle its own mess 😉

antimix, 2019-09-25:
Hi the original developer of SKINNYDIP has just updated his code, and implemented the request. 
Please go there : [https://github.com/domesticatedviking/PrusaSlicer/releases/tag/2.1.0-Skinnydip-Fast-1.0B](https://github.com/domesticatedviking/PrusaSlicer/releases/tag/2.1.0-Skinnydip-Fast-1.0B)
and post there any issue or development request.
Regards

lukasmatena, 2024-03-21:
A similar feature was implemented in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.

closed 2024-03-21 by lukasmatena