prusa3d/PrusaSlicer #2731: Reload from disk mispositions object below build plate

kevinkahn, 2019-08-08:
### Version
2.0.0+win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Mk3S

### Behavior
Just encountered an unexpected behavior doing a reload from disk that while perhaps not exactly a "bug" is close enough that it should get looked at.  I had a model that I had sliced and then decided that I wanted a slightly thicker bottom part.  So in Fusion I extruded the bottom a couple of additional mm and exported the new stl.  I then did a reload from disk and it seemed to work fine.  But then (after starting the print) I noticed that the bottom thickness was wrong.  It turned out that the model reloaded with the bottom of the model set that couple of mm below the print bed surface.  Slicing then ignored that part and I got what I had before.  (Actually worse because I had also moved a part of the top of the model down the same distance so I now had a thinner part between the bottom and the local "top" rather than the thicker part that I had wanted.  What is interesting is that it doesn't seem possible to purposely in PS move the model down such that it is below the print bed.  At the very least this is a completely unexpected behavior since it results in a slice that you could never purposely get.  I tried a simple test case with a 10x10x10 box that I then deepened to 12 high and got the same results.  

#### Project File (.3MF) where problem occurs
You can reproduce easily as I did with a simple box.  Load, slice, change box bottom position in CAD program, reload

Jebtrix, 2019-08-08:
Sounds like duplicate of issue [2541](https://github.com/prusa3d/PrusaSlicer/issues/2541)

kevinkahn, 2019-08-08:
Looks like it is a variant of the same issue.  I had done a search before opening but must have missed the previous issue.

bubnikv, 2019-08-09:
Closing this one as a duplicate.

closed 2019-08-09 by bubnikv