prusa3d/PrusaSlicer #2732: Include AppImage metadata

goncalossilva, 2019-08-09:
### Version
2.0.0

### Operating system type + version
Linux / AppImage

### Behavior
PrusaSlicer's basic AppImage support seems incomplete. There are a few problems:
1. If using [appimaged](https://github.com/AppImage/appimaged), the app has no icon. I presume it is missing one in the correct location. Cura seems to handle this [here](https://github.com/Ultimaker/Cura/blob/178f73ec620e08d4e3f7cd18a54981d242982f2a/CMakeLists.txt#L89-L90), similar to other apps. Looking at #451 it seems like this has worked at some point, but it's not showing for me on the latest version.
2. Similarly, it is not automatically associated with *.stl and other supported files. Cura also does this well, which I presume is a combination of [having a desktop file](https://github.com/Ultimaker/Cura/blob/master/cura.desktop.in) and [installing it](https://github.com/Ultimaker/Cura/blob/178f73ec620e08d4e3f7cd18a54981d242982f2a/CMakeLists.txt#L87-L88) during compilation. I presume the remaining 2 metadata files are useful, too.

I would be happy to dig further into this and try to contribute, but the process is undocumented as discussed in #1253.

bubnikv, 2019-08-19:
The icon problem is likely caused by the following issue:
https://github.com/prusa3d/PrusaSlicer/issues/1431#issuecomment-502438554

probonopd, 2019-09-21:
> Icon. If using appimaged, the app has no icon.

@goncalossilva please verify with the upcoming PrusaSlicer 2.1.0-beta.

> File associations. Similarly, it is not automatically associated with *.stl and other supported files. 

Cura has in its desktop file:

```
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;image/bmp;image/gif;image/jpeg;image/png;text/x-gcode;application/x-amf;application/x-ply;application/x-ctm;model/vnd.collada+xml;model/gltf-binary;model/gltf+json;model/vnd.collada+xml+zip;
```

PrusaSlicer 2.0.0 is lacking any `MimeType=` in its desktop file. Adding it there should do the trick.

bubnikv, 2019-09-23:
I have asked @lukasmatena to look into it. The purpose of MimeType is not
clear to me. What types should we put there for our slicer?

probonopd, 2019-09-23:
> The purpose of MimeType is not clear to me.

It allows the user to double-click, e.g., a `.stl` file, and it will be opened in PrusaSlicer.

bubnikv, 2019-09-24:
I was confused why Cura would want to register for image/bmp;image/gif;image/jpeg;image/png

probonopd, 2019-09-25:
Good question indeed. In fact I think it is not a good idea to do that.

goncalossilva, 2019-09-26:
There are quite a few discrepancies between what PrusaSlicer can open/import and what's in that list. This should probably be cleaned up and updated.

*.stl:
* `model/stl` ([official](https://www.iana.org/assignments/media-types/model/stl))
* `application/vnd.ms-pki.stl` (also used)
* `application/pla` (also used)

*.obj:
* ~`text/plain` ([apparently official](https://en.wikipedia.org/wiki/Wavefront_.obj_file), too broad to be used...)~
* `text/prs.wavefront-obj` (also used)

*.amf:
* `application/x-amf` ([apparently official](https://en.wikipedia.org/wiki/Additive_manufacturing_file_format))

*.3mf:
* `model/3mf` ([official](https://www.iana.org/assignments/media-types/model/3mf))
* `application/vnd.ms-3mfdocument` ([also official](https://www.iana.org/assignments/media-types/model/3mf))

I would suggest sticking with the "official" media types (except obj). That means PrusaSlicer would use:

`MimeType=model/stl;text/prs.wavefront-obj;application/x-amf;model/3mf;application/vnd.ms-3mfdocument;`

bubnikv, 2020-03-20:
I have updated the AppImage with 
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf; 
from Cura. It will be contained in the PrusaSlicer 2.2.0 final.

Closing.

closed 2020-03-20 by bubnikv

goncalossilva, 2020-06-08:
Thanks @bubnikv! Borrowing from Cura is a great improvement. I don't know if you saw the comment above, but not all ypes are up-to-date, nor are Cura and PrusaSlicer completely on par in what they can open and import. A better list that's tailored specifically to PrusaSlicer is:

**`MimeType=model/stl;text/prs.wavefront-obj;application/x-amf;model/3mf;application/vnd.ms-3mfdocument;`**

I'd be happy to PR this but it doesn't look like it's OSS. The reasoning is 2 comments above this one, and I'm happy to discuss further if needed. :+1: