prusa3d/PrusaSlicer #2733: Lost some small line

Thamone, 2019-08-09:
### Version
2.0.0

### Operating system type + version
Manjaro

### 3D printer brand / version + firmware version (if known)
Creality Ender5
### Behavior
I try to print [This Raspberry pi case](https://www.thingiverse.com/thing:3361218) It's looks normal when import but after sliced, it small line that connect grid with edge are lost. How to fix this?

#### Project File (.3MF) where problem occurs

[image]
[image]
[RPI-BOTTOM-8020_3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3485050/RPI-BOTTOM-8020_3mf.zip)

neophyl, 2019-08-09:
Turn detect thin walls OFF.  That seems to make it slice properly.

Thamone, 2019-08-09:
> Turn detect thin walls OFF. That seems to make it slice properly.

Thank you.

closed 2019-08-09 by Thamone