prusa3d/PrusaSlicer #2734: Using Layers editing does not trigger reslicing so the export G-code button remains

jokohoko, 2019-08-09:
### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows 10 Build 1809

### 3D printer brand / version + firmware version (if known)
All Prusa original machines

### Behavior
When changing settings the interface always updates to re-show the slice now button instead of the export g-code option. But when changing layer height using the Layers editing tool it does not. Thus the only way to reslice is to adjust other settings first.

[ Background processing is turned off ]

bubnikv, 2019-08-09:
I cannot reproduce it.

jokohoko, 2019-08-09:
Restarting my computer solved the issue. Unclear why, but thus probably just an anomaly.

closed 2019-08-09 by jokohoko