prusa3d/PrusaSlicer #2736: Slow dialog Multiple instances

thesalzar, 2019-08-09:
### Version
2.0.0+win64-201905201652

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win10 Pro Insider Preview - 18956.rs

### Behavior
Open two instances of PrusaSlicer.
On the second instance, add object,slice, export g-code.
The save as dialog takes a long while to load.
Instance 1 runs like normal.

rtyr, 2021-02-03:
I am unable to reproduce it with 2.3.0. Do you still have this issue?

thesalzar, 2021-02-03:
I am unable to reproduce this issue on 2.3.0 and it has not reoccurred.

closed 2021-02-03 by thesalzar