prusa3d/PrusaSlicer #2737: Wipe Tower Error When Filaments Different Diameter

wsmagee, 2019-08-09:
### Version
2.1.0-alpha1 + win64

### Operating system type + version
Windows 10 - 64bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2
### Behavior
This is new in the 2.1.0-alpha1 release.

If you have the slicer using multiple filaments.  Mine is setup for the MMU2.  And have setup different filaments, and measured the actual diameter of filaments, and entered that value into the filament diameter.  ANY difference in diameter throws the below error.
"PrusaSlicer has encountered an error
The wipe tower is only supported if all extruders have the same nozzle diameter and use filaments of the same diameter."

This makes sense if the filament diameter were 1.75mm vs 3mm.  However I have very minor differences in diamater like 1.75 vs 1.745.  It seems like there should be a threshold before this warning appears.

#### Project File (.3MF) where problem occurs
No specific file.

tkircher, 2019-08-09:
Does each filament not have an option for its own extrusion multiplier?

wsmagee, 2019-08-10:
Yes, however I have been measuring the average diameter of the filament and using that value in the filament size setting.  Then letting it adjust the flow per the actual size variance.  I only use the extrusion multiplier if I am trying to adjust based on the actual performance of the filament.

tkircher, 2019-08-10:
The filament sizes are nominal. So your 1.745mm diameter filament is really 1.75mm filament with some variance. The extrusion multiplier is intended for fine adjustments like that. If you are having problems with your prints, the best solution is to buy better filament, or invest in a filament width sensor.

wsmagee, 2019-08-10:
I understand the difference between the nominal size and actual.  And I'm not having any issues with my prints.
I just wanted to report this as with this version change, the behavior is now different.  Since there are nominal sizes like 1.75mm and 3mm then it's odd that the field allows for such small variance adjustments, then it reports it as a problem.  If this is the new intended behavior, so be it.  But it would be more logical to have that field have dropdowns with the most common standard sizes, and still allow for free entry (future proof).  But something different to make it more clear this is to select the nominal filament size moving forward, not record an average size for your particular roll of filament.

lukasmatena, 2019-08-15:
I'm responsible for the change, my logic was similar to what @tkircher says, i.e. diameter is nominal and variations should be dealt with by setting the extrusion multiplier. I did not realize that our own tooltips encourage users to do otherwise. Having the diameter as a dropdown with possible text input  would make more sense to me, but the behaviour chosen in the past is indeed different. Therefore, to not break existing profiles with slight variations in diameter, I allowed for a 10 % difference in filament diameter (https://github.com/prusa3d/PrusaSlicer/commit/15744f021a0aa84e3386fb8579cc47e43e427eb5). That should be more than enough.

Actually, the diameter is a floating point number so the test for strict equality could have been a problem regardless. Thanks for the report.

wsmagee, 2019-08-16:
I agree 10% variance will be plenty.  Thanks.

lukasmatena, 2019-08-20:
The above mentioned fix was merged into master, following release will have the 10 % tolerance. Thanks. Closing the issue.

closed 2019-08-20 by lukasmatena