prusa3d/PrusaSlicer #2740: [Feature Request] Add Filament Properties to Layers Editing

th33xitus, 2019-08-10:
### Version
PrusaSlicer-2.1.0-alpha1

### Request
Add Filament Properties as editable options in the Layers Editing Menu.
It would be handy to sometimes change temperature at a given height. 

I suggest adding at least the following options to be editable via Layers Editing:
- Extrusion multiplier
- Extruder temperature
- Fan speed
- Filament overrides (at least the retraction and wipe settings there)

Maybe this would be also very suitable to implement in modifier-settings as well.

albert2004, 2019-08-21:
+1 for this idea.
Changing filament temperature would simplify, for example, generation of temperature towers. From request, one important setting is missing:
- hotbed temperature

ProfEngr, 2020-01-05:
+1 for per-layer retraction option. This is the one thing I have to keep Simplify3D around to do, retraction tuning. Currently, S3D isn't playing nice w/ extrusion multipliers and PS2.1 is spot on.

rtyr, 2020-09-11:
This request is also included in your https://github.com/prusa3d/PrusaSlicer/issues/3635. Closing this.

closed 2020-09-11 by rtyr