prusa3d/PrusaSlicer #2741: Importing profile corrupts bed preview

GWdd, 2019-08-10:
### Version
3.8.0-rc

### Operating system type + version
Win 7 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Imported a v3.7.1 MK3 profile with modified start gcode.  Plate images now default to basic 250x210 sheet, not the "pretty" Prusa preview sheet.

The preview is hit or miss dependingon how the app is started. When using associaction (double click an STL the preview is generally wrong.  Open the app then import an STL then it sometimes works unless the STL was created by 3.8.0.

[image]

#### Project File (.3MF) where problem occurs
[Box Test.zip](https://github.com/prusa3d/PrusaSlicer/files/3488705/Box.Test.zip)

bubnikv, 2019-08-19:
> Imported a v3.7.1 MK3 profile with modified start gcode

Where did you get the profile from?

> Open the app then import an STL then it sometimes works unless the STL was created by 3.8.0.

Where did you get the 3.8.0-rc version or 3.8.0.?

GWdd, 2019-08-19:
The 3.7.1 is a mistake - sorry - wrong brain attached.  The custom profile was created by modifying the default, shipped with P.Slicer 2.0.0 MK3 profile, and saving it as a new user profile.   To import the profile, I tried a variety of methods, simplest was the Import Config and navigated to the <user>/Apps/Roaming/Prusa/PrusaSlicer/printer ... profile folder, selected the file, hit enter.

I suspect the file may be malformed as created by v2.0.0 since I've read similar sentiment about having to remove 3 lines before importing configs.

GWdd, 2019-08-19:
Here's the printer config.ini:
[PrinterProfile.zip](https://github.com/prusa3d/PrusaSlicer/files/3517149/PrinterProfile.zip)

Here's what I see after importing the printer config: all three profiles show a new progfile, but have .ini showing.
[image]

And here's what I see if I quite v2.1.0-alpha and open it again:  no new profile.
[image]

bubnikv, 2019-08-20:
There are two configuration types you may import / export:
Config
and
Config Bundle

The Config Bundle contains all the user profiles, and if imported, they are permanent. The Config contains just the active profiles and if loaded, they are temporary. If you want to make them permanent, you have to save them one by one.

GWdd, 2019-08-20:
Lack of documentation (aka User Manual) bites again.

closed 2019-08-20 by GWdd