prusa3d/PrusaSlicer #2742: Fill angle changes regards object orientation

dst74, 2019-08-10:
### Version
2.0
2.1 alpha

### Operating system type + version
Win10 Pro 1903

### 3D printer brand / version + firmware version (if known)
mk3 2.2.6

### Behavior
Fill angle 45.
Object rotate -45 on Z.
Fill is not anymore at 45 repsect to the object but it is at zero angle.

jokohoko, 2019-08-12:
Personally this is a feature I prefer since the fill angle is mostly a feature to speed up printing on the infill (printer can move faster under 45° than straight).
So i would not declare this a bug but just a practice in 3D printing settings that goes way back.

th33xitus, 2019-08-12:
Afaik the infill angle is always oriented relative to your printbed/origin. And i think it has been like this since.... forever? Other slicers do it the same way.

So not a bug at all. Simply adjust your infill angle for an orientation like this.

lukasmatena, 2019-08-12:
I'm closing the issue as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1529. The discussion may continue there. I personally don't have a strong opinion on that. Since apparently some consider it a bug and some a feature, it's probably best to leave it as it is.

closed 2019-08-12 by lukasmatena