prusa3d/PrusaSlicer #2744: [Feature Request]: Add Z-offset filament override

bortoni, 2019-08-10:
### Version
PrusaSlicer-2.1.0-alpha1

### Request
Since Filament override was added for retraction, it would be very useful to also add Z-offset.
This is currently only in Printer Settings.

PETG for example benefits greatly from being printed "a bit higher" off the bed vs PLA.

langph, 2019-10-24:
Agreed, and it would be the best to add the delta, not the actual offset stored in the printer for each steel sheet.

pboucher14, 2019-12-24:
That would be a good feature to add. Really usefull for PETG

mgg4, 2020-04-08:
I agree completely.  +1 for including this.  I'm currently using PrusaSlicer 2.2.0, and this feature would be very much appreciated.

Pelado-Mat, 2020-06-11:
+1. 

> Agreed, and it would be the best to add the delta, not the actual offset stored in the printer for each steel sheet.
The delta would be even better.

murphydan, 2020-07-19:
I was hoping something like this already existed for PETG. I add my vote!

userosos, 2020-10-29:
What about it? I cant see it the new build with alpha version.  I think that in the menu settings need add all of them settings from Print Settings and Printer Settings. It will be good!

skorokithakis, 2020-11-09:
There already is a "Z offset" setting in the printer tab, can we get a "Z offset" filament override? This ticket has been open for more than a year, it's a very useful feature that seems lilke it'd be simple and worthwhile to add.

DatenThielt, 2021-01-01:
Also would like to see this feature,  I have 3 different PLA suppliers and one of them requires more squish than the others,  as well as PETG liking a bit more height than PLA.  Easier to set it here than on the printer and "forgetting" the last height set 👍

JBFUK, 2021-01-10:
+2

JAYTEEAU, 2021-02-15:
+1
New to Prusa Slicer. This would be very useful indeed. Different materials would benefit greatly from this. IMO a perfect Live Z shouldn't need to be noted and changed per filament requirements

TheAtelier, 2021-03-03:
+1
This ticket has been opened for almost 2 years now.

1activegeek, 2021-05-19:
Coming to add a +1

@pboucher14 I see your reference here to some code - but I don't know much about PrusaSlicer code, so I'm not sure if that is just a skeleton fro the option or if that is the full bit of code needed for a pull request?

Alpacad, 2021-09-19:
+1

Coming from PrusaSlicer-2.4.0-alpha1 and found this while searching for the same feature requested idea.
Strange that this is still open after 2 years, having Z Offset in Filament Overrides would be a time saver when switching repeatedly from PLA to PETG and TPU.
Needless to say the times I forgot to adjust it in Printer Settings.

skorokithakis, 2021-09-19:
Good thing you commented, I was trying to find this comment for ages but failed. A workaround for this (for people who have autoleveling) that I do is to send `M851 Z<offset>` to get my printer leveling at a different height.

ardichoke, 2021-09-21:
Came here from the above issue that got marked as a duplicate. I'd also love to see this, or even better, if you could arbitrarily add any print settings to a filament profile with an override (I think ideamaker does something like this).

As a workaround, adding something like this to the filament start g-code has worked for me: ```
G0 Z0.2
G92 Z0.0```

This would add a 0.2mm positive Z offset to whatever the normal offset is.

Alpacad, 2021-09-21:
@skorokithakis as you say, this works for printers with auto bed levelling, to this day a lot of printers don't have a levelling probe as it is easier to setup.

@ardichoke this looks like a good compromise, but the custom filament g-code is inserted after any printer start g-code, sometimes this gives petg enough time to create a mess while priming the nozzle before starting to print.

skorokithakis, 2021-09-22:
Actually, @ardichoke's solution is better than mine, nice!

userosos, 2021-10-16:
> Came here from the above issue that got marked as a duplicate. I'd also love to see this, or even better, if you could arbitrarily add any print settings to a filament profile with an override (I [...]

Hi! What if i need Z offset set to -0.1 mm?

Alpacad, 2022-02-07:
In case it can be useful, I modified my filament start gcode.
@ardichoke gcode works very well, but I find that **the use of _placeholders_ and _macros_ describes better the process** of adding an offset. It makes it more "readable". 

So, I edidted my custom filament gcode like this:
; PETG Z offset
G1 Z{first_layer_height + 0.2} ; move nozzle to offset
G92 Z[first_layer_height] ; set current Z position as first layer height

Similarly to @userosos request, the same could be done for negative offsets:
; Negative Z offset
G1 Z{first_layer_height - 0.1} ; move nozzle to offset
G92 Z[first_layer_height] ; set current Z position as first layer height
This one should be even more simple than the previous method.

jremen, 2022-05-07:
+1

This feature is tremendously needed for PETG and TPU as well.

dougshell, 2022-08-08:
This would be helpful for ASA as well due to the increased bed temp

tommylux, 2022-11-27:
> In case it can be useful, I modified my filament start gcode. @ardichoke gcode works very well, but I find that **the use of _placeholders_ and _macros_ describes better the process** of adding an [...]

Neat workaround, but if you're like me who has multi extruder printer, the Filament Start G-Code is executed at every tool change. I need to ensure that it is only executed once (on the first layer. Tried the following logic:
`{if layer_num ==0}
G1 Z{first_layer_height + 0.2} ; move nozzle to offset
G92 Z[first_layer_height] ; set current Z position as first layer height
{else}
{endif}`
But it looks like layer_num is available in the current version of 2.5.0
[image]

A new filament override z offset setting would be nice.

evilC, 2023-04-13:
+1 vote for this feature.  
Yeah you could do it via gcode, but that's not exactly great UX

stefanct, 2023-05-20:
@rtyr I am tagging you because you are the only contributor having some interaction with this bug.
can you give us some insights on any prusa-internal discussion regarding this issue, if any? are there any implementation peculiarities that would make it hard to add?

Silvenga, 2024-06-08:
Note that `first_layer_height` may not actually be `0.0`. With built-in profiles `first_layer_height` is `0.2`.

drbobbob, 2024-07-01:
I believe babystepping is also be a viable option on the MK4/XL.  The buddy specific gcode page claims that it supports M290 and I tested with my machine that positive values move the nozzle up and negative values move it down.  I've been having good success with using the LCD to set a negative 0.010 mm offset so I'm adding this to my filament specific gcode:

`M290 S-0.010`

drbobbob, 2024-07-01:
I verified in the buddy firmware that the screen you get when you long press while a print is going also is also [invoking the babystep functionality](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/v6.0.3/src/gui/dialogs/liveadjust_z.cpp#L103)  Adding the `M290` code to the filament does not update that UI as it keeps track of its own offsetting.  That UI should still function, just not be updated by the manual gcode change.

According to the documentation the MK3 board does not support the `M290` babystepping command.  I think for that board either lying about the Z position (using the gcode that @Alpacad suggested) or using the probe offset (although that would be messy for owners of multiple print sheets if I understand how the `M851` code works).

I verified that the buddy firmware understands babystepping by running this gcode:
```
M290 S10
G4 P10000
M290 S-10
G4 P10000
```

It moves up 10mm, pauses for 10 seconds, moves down 10mm, pauses for 10 seconds and then is done in theory .  I ran this while the hot end was well away from the bed because I also saw that the direction for the babystepping could be reversed when configuring the firmware.

gaminnick, 2024-12-30:
+1

Does saying +1 get this moved up or is there something else that needs to be done to get this going?  This would be very nice to have and I imagine fairly easy.

gaminnick, 2024-12-30:
Would this

; PETG Z offset
G1 Z{first_layer_height + 0.2} ; move nozzle to offset
G92 Z[first_layer_height] ; set current Z position as first layer height

be different than
G1 Z0.2
G92 Z0.0

With a first layer height of 0.2 the first way would effectively be
G1 Z0.4
G92 Z0.2

Would this not start it higher than intended?

mpearson, 2026-09-29:
Per-filament Z-offsets would be extremely helpful for printing PHA, which requires NO bed heat at all. First layer calibration on my MK3S+ is only valid when the bed is hot so I have to manually apply the offset to the printer, even though it's a filament-specific issue. What would be *ideal* is if you can override arbitrary settings in the "custom parameters" box on the filament settings. I tried adding this JSON:
```json
{"z_offset": -0.15"}
```
but it did nothing. IMO these parameters should override existing parameters with the same name.