prusa3d/PrusaSlicer #2746: Using Autospeed and Gap fill is disabled

piyr9, 2019-08-10:
### Version
Prusa Slicer version 2.0.0+win64

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
hypercube with bowden on Duet Wifi

### Behavior
While using Autospeed settings like the following picture:

[image]

Results in missing Gap fill parts for example in the smoke stack in this Benchy:

[image]

Disabling autospeed gives the proper result like this:

[image]

When printing the model with Autospeed there are gaps in the print. If I set Gap Fill setting to 30mm/sec the model is filled in properly but Volumetric extrusion/Autospeed function is disabled.

zgrozemag, 2019-08-11:
Now if you would just read the tooltip, you'd know that setting gap fill turns gap fill off instead of using autospeed.
[image]

piyr9, 2019-08-11:
I have read the tool tip and I did set gap fill to 30mm and autospeed would be off printing really slow.

neophyl, 2019-08-11:
From your settings picture above you have Max Volumetric Speed set to 2.7mm, that's going to slow everything right down anyway.   You may alrerady be familiar wityh Max Volumetric but its an over ride on all extrusion speeds as you are basically saying my extruder cant supply enough hot plastic above this rate.  As such it applies a brake to all the printing speeds.  Thats why it can be set in both the Print profile and the Filament profiles, with whatever is the least being used.  Basically the Max Volumentric on print is normally set to match whatever your extruder can handle (around 11.5 for a E3 v6, more for a Volcano etc)., and if your filament is less then you set a lower level lime the PET filaments by defualt have (8mm).  As you have a Hypercube with bowden I dont know what a suitable value for your setup is.

It is a little weird that most speeds when set to 0 enables the autospeed setting but that for Gap Fill Speed it turns gap fill off but thats the way it is and at least theres the tool tip to let you know that.
Thats not a bug.

piyr9, 2019-08-12:
I wanted to print at 30mm/sec at a 0.2mm layer height with an extrusion width of 0.45mm. According to the volumetric calculator, it shows a 2.7mm3/s flow rate. This flow rate works and prints at 30mm/sec with gap fill disabled. It does not work with gap fill enabled.

[image]

oliof, 2019-08-26:
I am encountering this issue as well on PrusaSlicer 2.10alpha1 on linux. And, conversely, if gapfill is set to any speed while autospeed is on for every other speed setting, the remainder of the model is printed at a very low speed (7.2mm/sec) according to the PrusaSlicer preview. If I set gapfill to an "unreasonable" value like 120mm/sec, the PrusaSlicer preview shows that all features are printed at the lowest speed, except for gap fill which will be printed with the set value.

bernhardberger, 2020-02-18:
I can confirm this is still an issue in the current 2.2.0-alpha4.

Makes using autospeed basically useless :/

sleeperninja, 2020-04-07:
Auto Speed still useless in version: 2.2.0+

bubnikv, 2020-04-08:
Autospeed was always useless.

oliof, 2020-04-08:
@bubnikv autospeed works reasonably well when you disable gapfill, so I respectfully disagree. But if you feel it always was useless, maybe remove it?

bubnikv, 2020-04-08:
@bubnikv <https://github.com/bubnikv> autospeed works reasonably well when
you disable gapfill, so I respectfully disagree. But if you feel it always
was useless, maybe remove it?

Some features are kept for the compatibility with the upstream slicer,
though the two applications diverged a lot, so we may afford to remove some
legacy functionality. IMHO the autospeed is not very useful, so I would
tend to remove it one day. On the other hand, removing functionality is a
sensitive topic, some tweakers certainly find the feature immensely useful
and they will let the world know.

bernhardberger, 2020-04-17:
I have to admit that I had gotten by far the best prints/surface quality on symmetric objects when I played around with autospeed.

However missing gap fill and having holes in the object was a deal-breaker so I don't use it anymore :/

blalor, 2020-04-29:
I also ran into this problem. If gap filling needs to be disabled (why would you want to do that?), maybe that should be a separate checkbox, as `0` takes on too many meanings, and you can't leave gap fill speed empty.  Otherwise Auto Speed should fill gaps using a speed calculated from the max volumetric speed, or use the gap fill speed provided.

Honestly, it feels like autospeed should be a separately-enabled feature (like with a checkbox) with per-feature speeds overriding the calculated speed.

markd15, 2020-05-15:
I've been fighting with autospeed for the last few days and what I found is that you can turn on gap fill, but you need to crank your "Max print speed" setting to a really high value. Doing this and checking the result in the preview window will allow you to adjust the max speed of the print until everything is actually hitting your desired top speed.

Say you want to have a max autospeed of 60 mm/s but you need to have gap fill enabled. Enable gap fill with some reasonable value and then set max print speed to 365 mm/s and the actual max print speed will be 59.9 mm/s. Note that this is for my print profile specifically. This number changes dramatically with profile setting changes like layer height.

Now the problem with this method is the slicer still seems to think it's doing something right and will scale the speed of each line type accordingly (assuming you have different extrusion widths set for each line type). While the top print speed may be 60 mm/s, that may only apply to top surface lines and not infill lines, for example. 

I wish autospeed would just work perfectly with any layer height, line width, volumetric flow rate, or gapfill setting. Seems like some digging is needed to figure out what wires are twisted in the code that make gapfill screw with autospeed so badly.

bernhardberger, 2020-05-15:
I think the reason is as simple as: slow gap fill speed and long travel distances means that the extruder pressure can't be kept steady so it has to slow everything down.

Imho the "easy" fix (at least in theory) would be to just exclude gap fill from the equation for print speed calculation. I'd rather have a working autospeed for 99% of my print than having no gap fill at 100% of my print.

I personally think that autospeed is an underrated feature. Sure, LA (resp. PA) help a lot with keeping the extruder pressure steady - but I still had my best prints ever on some autospeed tests.

hannemann, 2020-05-24:
@markd15 Thanks for the workaround. I cranked up my max_print_speed by a factor of 6 and speeds seem to make more sense now. At least for me that works for now until this issue gets fixed.

sruggier, 2024-09-22:
> Imho the "easy" fix (at least in theory) would be to just exclude gap fill from the equation for print speed calculation. I'd rather have a working autospeed for 99% of my print than having no gap fill at 100% of my print.

The easy fix you suggested is actually possible in PrusaSlicer right now: if you set all of your speeds to some desired max value (like 200), _and_ you set a maximum volumetric flow rate, the slicer will use the fastest speed possible that remains below both of those maximums, even if gap fill is enabled.

If you really want the same flow rate for the entire print, you can make a second print profile with gap fill disabled and autospeed enabled, use it to see what speed the slicer picks, and then use that as your max flow rate on the print profile you actually print with.

The flow rate and print speed views in the slice preview are also very useful troubleshooting tools, to figure out exactly what part of your print is forcing the slicer to lower the overall flow rate. For example, if Arachne is enabled, there may be some very thin perimeters that hit the maximum speed instead of the maximum flow rate because of how thin they are.

Once you start configuring your prints in this way, you can experimentally find safe limits for your printer's speed, acceleration, flow rate, layer cooling time, etc., and the slicer will automatically print as fast as possible, without sacrificing quality.

sruggier, 2026-09-02:
I'm going to guess this issue didn't simply fix itself.