prusa3d/PrusaSlicer #2748: Cannot slice STL, Prusaslicer crashes

mickeysceme, 2019-08-12:
### Version
_2.0.0 Win64_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Win10 64 bit_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

###Prusa mk3s

### Behavior

Software crashes during slicing

# Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

mickeysceme, 2019-08-12:
[water_dripper_2.zip](https://github.com/prusa3d/PrusaSlicer/files/3492479/water_dripper_2.zip)

lukasmatena, 2019-08-12:
I tried in 2.0.0 and did not get a crash, but you are correct that it is not handled correctly - the bottom of the object is not sliced and the resulting empty layers confuse the slicer. It is quite possible that it really crashes on a different PC, because this is unsupported scenario.

The reason why the bottom was not sliced in the first place is most likely a broken model - I would try fixing it through NetFabb (should be possible from the context menu on Win10).

I also recommend to try out recently released PrusaSlicer2.1.0-alpha1. This version would emit an error message about what went wrong and would stop slicing on its own.

mickeysceme, 2019-08-12:
Hi Lucas,

Ir was slicing at 30 percent or si, than crashed, every time. I'll try the 2.1 alpha. 
Thanks

jokohoko, 2019-08-12:
This model is just a nightmare to print. 
You have a bunch of tubes in the bottom that are only 0.2mm thick. 
Therefore the slicer ignores these features and the software does not calculate them. (and zooming to 30% would only make this worse)
In my case both V2.0.0 and V2.1.0 are able to slice the part and show correct warnings about having an empty first layer. Though I would never print a file like this without massive modifications.

mickeysceme, 2019-08-12:
I see. I've downloaded it from thingy..... I managed to fix it, but still.... Nightmare

closed 2019-08-12 by mickeysceme