prusa3d/PrusaSlicer #2751: 2.0 Error Exporting Config Bundle

AbeFM, 2019-08-12:
### Version
2.0, Win10

I exported a config bundle in my release version of 2.0, when I tried to load the bundle into 2.1a, I got the following error:

[image]

Looking at the export, checking the specific line, I found that to be the last in the file:

[image]

From this I infer my export didn't work right. Curiously, PS2.0 exits during/after the config bundle export, perhaps it is crashing, I do not know its intended behavior. It indicates it is exporting, then blinks out of existence after a short pause.

The first time, I did it with a temporary profile in memory from a project I had just loaded. The second time I did it clean, both times the error was on the same line, but this may make sense if it doesn't add the temporary one to the bundle. It always crashes with the same incomplete file, this may change with different profiles installed, I don't know.

Jebtrix, 2019-08-12:
Export config bundle was indeed crashing in 2.0.0 leaving unfinished files. It is fixed in 2.1.0alpha1. 

> Alpha Changelog: Fixed crash when exporting a config bundle in case no SLA printer profile was activated yet #2307 #2336 #2370 #2446 #2483 #2611.

AbeFM, 2019-08-12:
Sounds like you can close it? 
Is there a work around? I'm glad it was fixed and that may be all that can be reasonably done - but not being able to bring my old profiles over is making life harder.

It may be that exporting each profile/filament/etc individually is that answer, but it would be good to know. Do you have any advice? from the above, it sounds like *adding* an SLA profile under 2.0.0 will allow me to export?

Jebtrix, 2019-08-12:
Yes, if you add the SLA printer and then select it as your active profile _**once**_, then you can select your actual printer and do the export config bundle. 
Otherwise you can just manually copy the profiles, ie on windows "C:\Users\<usernamehere>\AppData\Roaming\PrusaSlicer". Just replace/add the \filament, \print, and \printer folder to "C:\Users\<usernamehere>\AppData\Roaming\PrusaSlicer-alpha"

lukasmatena, 2019-08-12:
I'm closing this issue, the bug has been fixed in 2.1.0-alpha1 and the problem of exporting config bundle from the buggy 2.0.0 has workarounds that have been described here (and in the other issues that @Jebtrix mentioned, e.g. https://github.com/prusa3d/PrusaSlicer/issues/2307#issuecomment-517357752).

closed 2019-08-12 by lukasmatena

AbeFM, 2019-08-12:
> Yes, if you add the SLA printer and then select it as your active profile _**once**_, then you can select your actual printer and do the export config bundle.
> Otherwise you can just manually copy [...]

FYI: I just added the printer through the wizard (unchecked everything else). I got a file 1 kbyte larger (168 vs 167), but it worked and included all my printers. I did not have to select mine, which is good because there are several of them.

> I'm closing this issue, the bug has been fixed in 2.1.0-alpha1 and the problem of exporting config bundle from the buggy 2.0.0 has workarounds that have been described here (and in the other issues that @Jebtrix mentioned, e.g. [#2307 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/2307#issuecomment-517357752)).

Sounds reasonable to me. :-)

Goury, 2019-08-27:
Workaround doesn't work on Ubuntu for me