prusa3d/PrusaSlicer #2752: Enabling supports causes an error that prevents gcode generation

liftbag, 2019-08-12:
### Version
_PrusaSlicer Version 2.1.0-alpha1+_

### Operating system type + version
_macOS Mojave_
_version 10.14.6_

### 3D printer brand / version + firmware version (if known)
_not relevant_

### Behavior
* _enabling support and auto generated supports or support enforcers, causes an error that prevents gcode generation_
* _enable "Generate support material"_
  _enable "Auto generated supports"_
  _Click on "Slice now"_
* _Expected Results: gcode is generated_
* _Actual Results: PrusaSlicer error window is opened as you can see in the attached screenshot_
[image]

_This is not a new feature request_

#### Project File (.3MF) where problem occurs
_Seems a general issue. Any project I've tried has the issue._

[loubie_aria_dragon.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3494339/loubie_aria_dragon.3mf.zip)

lukasmatena, 2019-08-12:
Thanks for the report. The test for empty layers does not deal well with the negative support contact z distance. We'll fix this.

lukasmatena, 2019-08-13:
This should be fixed by https://github.com/prusa3d/PrusaSlicer/commit/c18914022147ad0edb3c12ad1011a158d9c1e9eb. Thanks again for the report.

However, I noticed that an assertion fails when slicing your 3MF. This may not be serious, but we should at least check what's going on. Could you please keep the issue open for now as a reference?

`EdgeGrid.cpp:801: void Slic3r::EdgeGrid::Grid::calculate_sdf(): Assertion det != 0' failed.`

liftbag, 2019-08-13:
Sure. I'll keep open for your evaluations.

lukasmatena, 2019-08-20:
I'm closing the issue, it should be fixed. The problem with the assert is unrelated and the reason is known to us.

closed 2019-08-20 by lukasmatena