prusa3d/PrusaSlicer #2753: modifier shape with fill not working correctly

sbierly, 2019-08-12:
### Version
2.0.0+win64 201905201652

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 latest updates

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 

### Behavior
Added a cylindrical modifier shape. Put it in the center of the shape, fully overlapping footprint, only covering the Z midsection. Added Infill modifier. Changed fill to 100% (primary setting was 60%). Changed to Grid (primary was gyroid). Sliced. Midsection (covered by cylinder) changes to Grid, fill remains at 60%. 

_Is this a new feature request?_
No, bug.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

sbierly, 2019-08-12:
I would upload the .3MF but see no way to do so.

Jebtrix, 2019-08-13:
You just have to zip it.

Jebtrix, 2019-08-13:
Actually without seeing your file its because grid is a sparse infill method. Try setting your model to %100 grid infill in global Print Settings. It won't let you. The modifier is bypassing that message but still refusing to do it. The most is %90.

sbierly, 2019-08-13:
Hmmm, I guess that explains why when I ripped everything up a 3rd time and
played with it I got it to work in Rectilinear. In over my head I guess, I
don't even know the difference, the messages in the top mode were keeping
me in line and I didn't care enough to remember. I picked grid thinking
that was the right one...might be nice to have those warnings in the
modifiers too, but I realize you guys added a cool-TON of stuff and it's a
lot to keep track of. Extremely cool stuff, thanks!
--Scott

sbierly, 2019-08-13:
[2nd Furniture extension round 15mm v2.zip](https://github.com/prusa3d/PrusaSlicer/files/3495093/2nd.Furniture.extension.round.15mm.v2.zip)

I suppose this is resolved, but here is that Zip.

Jebtrix, 2019-08-13:
I agree the modifier should invoke the warning too.

YuSanka, 2019-08-14:
Thanks for your report. We planing to add option updating as it's implemented for "global settings". Working on it in [SPE-1002](https://dev.prusa3d.com/browse/SPE-1003)

bubnikv, 2019-08-23:
The fix will be in the upcoming PrusaSlicer 2.1.0-beta. Closing.

closed 2019-08-23 by bubnikv