prusa3d/PrusaSlicer #2754: Layer time estimate in preview

rcmaniac25, 2019-08-13:
### Version
2.0.0+win64-201905201652

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
This is a feature request.

So I don't know what others do, but I tend to leave PrusaSlicer up and follow along with the print, picking a spot where geometry changes to a greater extent then before and double checking the printer handles that without issue. I realize this is a waste of time. "Trust the printer" (assuming you do trust it...)

I also do this for areas I'm worried about. Such as a particularly steep overhang that I hope will be created without too much drooping or similar.

Thought hit me that like #1621, what if you could get "time till a layer" using the layer sliders in the preview view?

Imagine it like this: bottom layer/Layer 1 says "0s" or "Start" or something. Layer 2 is "10s", Layer 3 is "12s", Layer 4 is "14s" etc.

Now, when I find that layer I'm worried about of the 16h print, I can see it says "10h 32m 6s" and can literally set a timer to come back and check instead constantly looking at the print. Or at least know if I can do something before a point where I want to watch in case I need to cancel or just want to stare at the printer for a bit.

Now, one way to make this even better: make it relative to the lower layer selector. Continuing the prior example:
I move the slider from Layer 1 to Layer 2: Layer 2 is "0s", Layer 3 is "2s", Layer 4 is "4s", etc.
I move the slider from Layer 2 to Layer 3: Layer 3 is "0s", Layer 4 is "2s", etc.
etc.

Sorry if a bit of rambling, let me know if any other questions or comments.

MrManny, 2019-09-04:
I wish to add my +1 to this feature request. It would also greatly help with manual filament changes using M600 codes, so I have a rough idea about when I should be around to do the thing.

lukasmatena, 2019-09-04:
@MrManny The time estimates for individial M600 toolchanges have been requested in https://github.com/prusa3d/PrusaSlicer/issues/2390 - it is already implemented and will be in the next release 2.1.0 (due in several days probably).
[image]

MrManny, 2019-09-04:
@lukasmatena, thank you. You are a gentleman and a scholar! I was unaware of this development, but it pleases me greatly to hear that.

bubnikv, 2021-01-13:
PrusaSlicer 2.3.0 shows print time to the end of a layer at the vertical slider.

rcmaniac25, 2021-01-13:
Ah, did not see that before before. Very cool. Not exactly what I was looking for, but good to know.

1. Just a comment on the existing implementation: it seems to be doing a text wrap, so it puts the print time and layer time next to each other (confused me at first... maybe not as obvious as it originally seemed) but then wraps if too long, which also threw me off until I realized it was wrapping the text.
2. The original goal was to figure out how long until a layer. So when starting a print, the existing implementation is good. But once N layers in, then it ceases to be relative and now it's still a guessing game of "how much longer till this layer?". As stated in OP, maybe it's just how I use it.

Drachenkaetzchen, 2021-02-23:
I I have added support to display the estimated layer printing time. The pull request can be found in #6115 .

[image]

closed 2026-09-29 by prusaslicer-bot[bot]