prusa3d/PrusaSlicer #2755: Slicer Overwrites accelleration and speed values

Louis-Cypher, 2019-08-13:
Dear all, 

I think I found a bug in the most recent version of this excellent software leading to printing problems on my Marlin based printer (I was about to file a bug report to Marlin but found it is the slicer who produces the Problem.)

### Version
2.0.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
reprap Marlin 2.0.x- Bugfix

### Behavior
Even though acceleration control "standard" is set to 0, slicer exports the following to the g-code:

> ; generated by PrusaSlicer 2.0.0+win64 on 2019-08-13 at 10:54:12
> 
> ; 
> 
> ; external perimeters extrusion width = 0.56mm
> ; perimeters extrusion width = 0.56mm
> ; infill extrusion width = 0.56mm
> ; solid infill extrusion width = 0.56mm
> ; top infill extrusion width = 0.50mm
> ; first layer extrusion width = 0.52mm
> 
> M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
> M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
> M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
> M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
> M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec

To my understanding this is a bug since Slicer should not overwrite my settings if set to 0. Thanks in advance for your support

[config.zip](https://github.com/prusa3d/PrusaSlicer/files/3496257/config.zip)

lukasmatena, 2019-08-13:
> Even though acceleration control "standard" is set to 0

I don't understand. What is set to zero and where? I cannot find anything in your config that would allow me to make sense of it.

Louis-Cypher, 2019-08-13:
You will not find it in my ini since it is the default setting. You'll find it under >Print Settings > Speed > Acceleration Control > Default (sorry had to change the menu language)

lukasmatena, 2019-08-13:
Ok, I understand now. You turned off acceleration control in Print Settings and are surprised to see an M204 command in the gcode.

The M201-M205 commands in the gcode that you posted are emitted at the beginning and can be configured in Printer Settings->Machine limits. Being sure about the limits is necessary for the time estimator  to work reliably.

If you for some reason do not want to have these in the gcode, just set gcode flavor to RepRap (it is only emitted for Marlin). Apart from this, the behavior should not change.

Louis-Cypher, 2019-08-13:
Since I use Marlin I'd prefer to stick to Marlin flavor. Sadly the behaviour rapidly changes. My z- axis stalls as soon as the g-code is loaded. Since I use a threaded rod the acceleration set seams to be much too high. My maximum acceleration is set to 100mm/s(strange unit for an acceleration!)  in Marlin. The M201 seams to set a too high acceleration limit. The M204 then sets a too high overall acceleration. I'd prefer to have a possibility to not overwrite my printer settings.

BTW: I am not in favour of defining acceleration settings that are dependent of the transmission ratio since it is a motor parameter, but I am fully aware that this is a marlin issue.

I have another issue with the introductory code. My printer starts at 5 mm distance to the build plate if I don't delete the line defining the rest position at z5

lukasmatena, 2019-08-13:
If the values are too high, I would set them lower. And yes, that requires editing the printer profile. Is there a reason to use a profile that does not match your printer? Alternatively, set gcode flavor to RepRap. As far as I know, the only difference is setting linear advance (M900 vs M572), but that is taken care of it filament custom start gcodes anyway. With a single exception - resetting linear advance on the wipe tower (which I assume you don't use). But then you won't get reliable time estimates.

I agree with you that this is not well documented and the various gcode flavors that PrusaSlicer supports do not make it any easier. It might make sense to allow disabling those commands.

> My printer starts at 5 mm distance to the build plate if I don't delete the line defining the rest position at z5

I don't understand.

Quas7, 2020-10-10:
as a workaround for that, I just added M501 in "Printer settings", "start G-code" to reload EEPROM machine settings and reset all slic3r changes.

lukasmatena, 2020-10-10:
@Quas7 Thanks for the tip. However, thie emission of the M203 etc. commands can be suppressed in the upcoming 2.3 release (see #1212). I believe this issue can be therefore closed.

closed 2020-10-10 by lukasmatena