prusa3d/PrusaSlicer #2756: Reload from disk duplicating objects

mike-berriman, 2019-08-13:
### Version
2.0.0+win64

### Operating system type + version
WIndows 10 x64

### 3D printer brand / version + firmware version (if known)
Original Prusa 2.5s unmodified

### Behavior
I have a 3mf which contains multiple instances of each of three .stl files.  If I select one instance of one stl and choose to reload from disk, it seems to reload that particular stl correctly.  However, it also imports copies of all instances of the other objects in the scene.  The newly imported objects are not positioned on the plate in any way (that I can tell) it just seems to put them in the scene anywhere.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[image]
[image]

Jebtrix, 2019-08-13:
There is definitely some unintuitive behavior in Reload Disk in 2.0.0 and 2.1.0alpha1. I'm assuming your 3MF file was from a PrusaSlicer saved project? Once its a project and the external file references are gone it gets odd. I tried to illustrate the behavior in a clip using two simple single part STLs. 

[image]

mike-berriman, 2019-08-13:
Yep - that's the behaviour I am seeing.  I'm not sure I'd call it unintuitive - I'd call it broken.  What is the purpose of the behaviour working as it currently does?

I didn't realise that the .3mf file lost reference to the original files.  That's a shame as that's really the behaviour I was after.  Is there functionality that stores the plate layout, without losing the file links?

Jebtrix, 2019-08-14:
Unintuitive was a bad choice of words. Predictably broke maybe? There is no external reference to the files in any export/save as options. PrusaSlicer should check if the filename exists in the same folder as the opened amf/3mf project but it does not.

Looks like these Reload Disk issues are getting some dust on it: [2526](https://github.com/prusa3d/PrusaSlicer/issues/2526), [1890](https://github.com/prusa3d/PrusaSlicer/issues/1890)

bubnikv, 2019-08-19:
Reload from disk was broken since 2.0. @enricoturri1966 will be fixing the feature for 2.2.

WarrenSchultz, 2020-02-28:
I believe this is broken in a different way now, unless I misunderstood the original bug.
Reload from disk of a single part will now prompt you if you want to reload as a multipart. If you select no, it will drop a stack of all the parts in the scene on top of the part you were trying to reload.

bubnikv, 2020-02-29:
Please review the "Reload from disk" in PrusaSlicer 2.2.0-rc and please close this issue if it works as expected. Thank you.

rtyr, 2020-05-27:
@mike-berriman Can you try it again with 2.2.0?

bubnikv, 2020-10-12:
No response, closing.

closed 2020-10-12 by bubnikv