prusa3d/PrusaSlicer #2758: [Bug Report] Changing of extrusion witdths not working on 2.1.0-a1

OBELIKS, 2019-08-13:
### Version
Version 2.1.0-alpha1+win64

### Operating system type + version
Windows 10 Pro 1903 64 bit

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S

### Behavior
It is not possible to change the any of the extrusion widths on 0.3mm DRAFT MK3 profile.
On a copy of the profile it is possible to change perimeter widths.
0.35mm FAST MK2.5S has the same problem, so possibly others have it too.

Jebtrix, 2019-08-13:
Works fine on Win7. I think this is a Win10 permissions issue. When you make a copy it goes into the permissive user profile \Appdata. Where did you extract the alpha version to? Try moving it somewhere else ie Desktop. It could be possible it doesn't have file permisison to edit ..\prusa-slicer.exe\resources\profiles\PrusaResearch.ini where defaults are derived from and it manifests as uneditable.

Jebtrix, 2019-08-13:
Oh wait do you mean save over the default profile or just temporary changes in Print Settings? Default profiles are reserved.

OBELIKS, 2019-08-13:
The problem is not when saving, it happens on changing the values, they just jump back when the field loses focus. When I changed them in the ini file they were read correctly.

OBELIKS, 2019-08-14:
I figured out how to replicate it. You can start with fresh profiles. First custom Print profile can be created without a problem, but when you try to create a second one, you will notice some values jump back to original, when the field loses focus.
This is NOT connected to a specific original profile.
If needed I can create a small clip that shows this.

YuSanka, 2019-08-14:
Thanks for your report, but we can't reproduce this issue. So, small clip can be helpful.

OBELIKS, 2019-08-14:
Here is the clip. I started with a fresh profile. I change some of the settings, save the first custom Print profile and then go to change another one. If you skip to around 1m30s you will see that the print speeds start to jump back. If I close PrusaSlicer and open it again it works again for a while.
https://drive.google.com/open?id=1mefW2MfAXvkn-2gJx9enRMukR2Fjn2fZ

YuSanka, 2019-08-14:
Thank for you report and help with provided clip. I believe this bug is already fixed with 08dced3ce7c28e9024abdf2ea732782b966af1d6 and will be a part of next release.

closed 2019-08-14 by YuSanka