prusa3d/PrusaSlicer #2760: [ESP600] command in Start or End G-Code (ESP3D)

DROID8622, 2019-08-14:
### Version
2.0.0+win64
### Operating system type + version
Windows 10 x64
### 3D printer brand / version + firmware version (if known)
Custom
### Behavior

I am trying to use [ESP600] command in the End G-code to send notification , but it says failed due to invalid custom G-code sections

lukasmatena, 2019-08-14:
If you want to add [ESP600] into the gcode and prevent the parser from evaluating the inside of the square brackets, you have to mark it as string, that is
`{"[ESP600]"}`

lukasmatena, 2019-08-19:
I added a mention of this construct in the respective [wiki page](https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language).

I assume this solves the issue. Closing.

closed 2019-08-19 by lukasmatena